logisim-evolution/logisim-evolution #1335: Feature Request: Allow the use of a custom shape for LED's in circuit appearance

flarn2006, 2021-12-20:
Currently, as far as I can tell, when you use the Dynamic Components button to show an LED on a subcircuit, it'll always appear as a circle, in a small size that cannot be changed. I propose that this be improved so that LED's (both normal and RGB) can be shown on the circuit's custom appearance in a user-defined shape and size, perhaps even as text that changes color. This would streamline the creation of essentially any type of custom segmented display by enabling it to be done in a regular circuit, without having to code a JAR library for it.

Potential further enhancements could include:

- Allowing 7-segment/hex displays to be customized further in circuit appearance, so they can be displayed in a custom size without the border.
- Allowing other types of LED displays to be included in circuit appearance with similar customization options (e.g. LED Bar, LED Matrix, and perhaps even RGB Video if it's feasible)
- When including dynamic components of a subcircuit, have them be automatically placed in the same arrangement as they are on the subcircuit's custom appearance. Even better, a new appearance element could be added which will display all dynamic components of a subcircuit as a single object, and a new property could be added to shapes and text to determine whether or not they will appear in this object.

maehne, 2021-12-22:
Thanks @flarn2006 for your enhancement proposal! @BFH-ktt1 may be able to provide some feedback how complex it would be to implement it. PRs in this direction are certainly welcome!

MinekPo1, 2025-11-03:
I don't have much experience with java but taking a look at some of the code I have a hope that I will be able to get at least some of this to work and I would like to see these features too, as long as I am able to gather the energy.  Also I hope my choice to add this comment to this issue rather than creating a new one was the right one, especially as some of the things I'm mentioning are beyond the scope of the original issue.

Just throwing out what I feel I could do:
- Adding a "scale" attribute for existing dynamic elements
- Copy over color settings from the circuit versions
- Maybe implement `(...).circut.appear.DynamicElementProvider` for the LED Matrix & LED array

Some questions I have lingering:
- Would remaining memory components benefit from implementing `(...).circut.appear.DynamicElementProvider`?  If so how would that look like for the RAM/ROM components?
- Maybe a component which displays a single character would be beneficial? Should be easy to implement (as a stripped down tty) and likely way more useful as a `(...).circut.appear.DynamicElementProvider` than a tty considering a tty needs two clock cycles per character
- Would the PLA component (the one in the gates cat.) benefit from being a `(...).circut.appear.DynamicElementProvider`?  I feel like this could be quite beneficial for user created circuts with complex behaviour controlled by a PLA (ex. an APU)

MinekPo1, 2025-11-04:
I'm struggling to do so unfortunately as it seems there has been a directory tree change between current logisim-evolution and holy-cross, we'll discuss further in the PR