logisim-evolution/logisim-evolution #1336: Nimbus timing diagram null pointer exception

koxx4, 2021-12-22:
In preferences choose Look and feel "**Nimbus**", after restarting app, opening a circuit and choosing timing diagram throws null pointer exception.

`Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException: Cannot invoke "java.awt.Color.getRed()" because "c" is null
        at com.cburch.logisim.gui.chrono.ChronoPanel.darker(ChronoPanel.java:313)
        at com.cburch.logisim.gui.chrono.ChronoPanel.<init>(ChronoPanel.java:67)
        at com.cburch.logisim.gui.log.LogFrame.<init>(LogFrame.java:199)
        at com.cburch.logisim.proj.Project.getLogFrame(Project.java:332)
        at com.cburch.logisim.gui.menu.MenuSimulate$MyListener.actionPerformed(MenuSimulate.java:338)
        at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1972)
        at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2313)
        at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405)
        at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262)
        at java.desktop/javax.swing.AbstractButton.doClick(AbstractButton.java:374)
        at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:1028)
        at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:1072)
        at java.desktop/java.awt.Component.processMouseEvent(Component.java:6626)
        at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3389)
        at java.desktop/java.awt.Component.processEvent(Component.java:6391)
        at java.desktop/java.awt.Container.processEvent(Container.java:2266)
        at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5001)
        at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2324)
        at java.desktop/java.awt.Component.dispatchEvent(Component.java:4833)
        at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4948)
        at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4575)
        at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4516)
        at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2310)
        at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2780)
        at java.desktop/java.awt.Component.dispatchEvent(Component.java:4833)
        at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:773)
        at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:722)
        at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:716)
        at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:97)
        at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:746)
        at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:744)
        at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
        at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:743)
        at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
        at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
`
_Checked on 
`openjdk version "17.0.1" 2021-10-19`,  Linux_

maehne, 2021-12-22:
I can confirm the bug. Nimbus look & feel triggers also on macOS the same exception when opening the timing diagram. Other look and feels seem not to be affected.

MarcinOrlowski, 2021-12-27:
The fix is simple but before I address this issue I need to understand **why** it occurs. If this occurs in most recent code, then it's because 

```java
selectBg = UIManager.getDefaults().getColor("List.selectionBackground");
selectHi = darker(selectBg);
``

and `selectBg` is returned `null`. `darker()` is not checking against null (and expect non-null argument). But why is `List.selectionBackground` not found in the first place?

MarcinOrlowski, 2021-12-29:
As guesses, there's no color definition we are looking for in this particular style:

```
key: List.actionMap value: javax.swing.plaf.basic.LazyActionMap@79c3e190
key: List.background value: DerivedColor(color=255,255,255 parent=nimbusLightBackground offsets=0.0,0.0,0.0,0 pColor=255,255,255
key: List.cellNoFocusBorder value: javax.swing.plaf.BorderUIResource@255b53dc
key: List.contentMargins value: javax.swing.plaf.InsetsUIResource[top=0,left=0,bottom=0,right=0]
key: List.disabled value: DerivedColor(color=214,217,223 parent=control offsets=0.0,0.0,0.0,0 pColor=214,217,223
key: List.disabledText value: DerivedColor(color=142,143,145 parent=nimbusDisabledText offsets=0.0,0.0,0.0,0 pColor=142,143,145
key: List.dropLineColor value: DerivedColor(color=115,164,209 parent=nimbusFocus offsets=0.0,0.0,0.0,0 pColor=115,164,209
key: List.focusCellHighlightBorder value: javax.swing.plaf.BorderUIResource@3c22fc4c
key: List.focusInputMap value: javax.swing.plaf.InputMapUIResource@420fe8be
key: List.focusInputMap.RightToLeft value: javax.swing.plaf.InputMapUIResource@7decaeb7
key: List.font value: javax.swing.plaf.FontUIResource[family=Noto Sans,name=Noto Sans,style=plain,size=12]
key: List.foreground value: DerivedColor(color=0,0,0 parent=text offsets=0.0,0.0,0.0,0 pColor=0,0,0
key: List.opaque value: true
key: List.rendererUseListColors value: false
key: List.rendererUseUIBorder value: true
key: List:"List.cellRenderer".contentMargins value: javax.swing.plaf.InsetsUIResource[top=0,left=0,bottom=0,right=0]
key: List:"List.cellRenderer".opaque value: true
key: List:"List.cellRenderer"[Disabled+Selected].background value: DerivedColor(color=57,105,138 parent=nimbusSelectionBackground offsets=0.0,0.0,0.0,0 pColor=57,105,138
key: List:"List.cellRenderer"[Disabled].textForeground value: DerivedColor(color=142,143,145 parent=nimbusDisabledText offsets=0.0,0.0,0.0,0 pColor=142,143,145
key: List:"List.cellRenderer"[Selected].background value: DerivedColor(color=57,105,138 parent=nimbusSelectionBackground offsets=0.0,0.0,0.0,0 pColor=57,105,138
key: List:"List.cellRenderer"[Selected].textForeground value: DerivedColor(color=255,255,255 parent=nimbusLightBackground offsets=0.0,0.0,0.0,0 pColor=255,255,255
key: ListUI value: javax.swing.plaf.synth.SynthLookAndFeel
key: List[Disabled+Selected].textBackground value: DerivedColor(color=57,105,138 parent=nimbusSelectionBackground offsets=0.0,0.0,0.0,0 pColor=57,105,138
key: List[Disabled].textForeground value: DerivedColor(color=142,143,145 parent=nimbusDisabledText offsets=0.0,0.0,0.0,0 pColor=142,143,145
key: List[Selected].textBackground value: DerivedColor(color=57,105,138 parent=nimbusSelectionBackground offsets=0.0,0.0,0.0,0 pColor=57,105,138
key: List[Selected].textForeground value: DerivedColor(color=255,255,255 parent=nimbusLightBackground offsets=0.0,0.0,0.0,0 pColor=255,255,255
```
Got no idea why and if that's their bug or it can be expected.

maehne, 2021-12-30:
@MarcinOrlowski: Thanks for your analysis and PR #1341 to fix this issue!

closed 2022-02-15 by koxx4