logisim-evolution/logisim-evolution #1338: Load file not work for the first output of RAM in command line mode

yhding5618, 2021-12-23:
The upper RAM is set to asynchronous read mode. So the output should change immediately after I load a new image to the RAM.

It works as expected in the GUI. The RAM and ROM always generate same output(_cmp_out_ always 1) when I load same image to them.

[image]

However in command line mode.

```
> logisim-evolution test-adder.circ -tty table -load ram.txt
0000 0000	0000 0000	1101 0000	0
0000 0001	1101 0001	1101 0001	1
0000 0010	1101 0010	1101 0010	1
0000 0011	1101 0011	1101 0011	1
0000 0100	1101 0100	1101 0100	1
0000 0101	1101 0101	1101 0101	1
0000 0110	1101 0110	1101 0110	1
0000 0111	1101 0111	1101 0111	1
```

These four columns are _adder_out_, _ram_out_, _rom_out_ and _cmp_out_.

The first output of RAM is always zero while the first output of ROM is as expected.

Content in _ram.txt_:

```
v2.0 raw
d0 d1 d2 d3 d4 d5 d6 d7
d8 d9 da db dc dd de df
```

Version:

```
Logisim-evolution v3.7.2
https://github.com/logisim-evolution/
master/26482a19 (2021-11-09T10:13:22+0100)
OpenJDK 64-Bit Server VM v17.0.1 (Oracle Corporation)
```

dtowell, 2022-07-18:
@yhding5618 can you post your .circ file? Anything come up since it was reported?

dtowell, 2022-07-19:
I can reproduce this "bug" in the GUI. I believe the problem is a misunderstanding about how RAM works compared to ROM. (I could be wrong, but this is my current understanding.) ROM always uses async operation (no clock is ever used). RAM defaults to use non-asynchronous reads, which means it must wait for the clock to raise (or lower, if you have set that option) before the output ports are valid. By changing the RAM attribute "Asynchronous read:" to Yes, you should get the results you expect.

@BFH-ktt1 Assuming I'm understanding correctly, I think this issue can be closed.

BFH-ktt1, 2022-08-31:
@dtowell : Yes, you are correct, the ROM's have only the asynchronous read mode, the RAM's, however, have synchronous and asynchronous read, where synchronous read is default. However, I'm not sure if it explains the problem.

ChristopherPAndrews, 2023-01-24:
It appears that this issue has been sitting in limbo for a few months. I found this because I just bumped into this same problem. I can confirm that this is _not_ because the the RAM is in synchronous read mode. I made a similar test circuit and set the RAM to be in asynchronous read mode. 

[image]

When the memory file is loaded, the initial output is 01 (the first value in the file). Stepping the clock walks through all of the values correctly.

When I run on the command line: `> logisim ram_test.circ -tty table -load simple_count.txt`

I get this:
```
0000 0000
0000 0010
0000 0011
0000 0100
0000 0101
0000 0110
0000 0111
0000 1000
0000 1001
0000 1010
0000 1011
0000 1100
0000 1101
0000 1110
0000 1111
1111 1111
```

The first line of the output should be `0000 0001`, but is is `0000 0000` instead. 

If the RAM is left in synchronous read, then the output is

```
EEEE EEEE
0000 0001
0000 0010
0000 0011
0000 0100
0000 0101
0000 0110
0000 0111
0000 1000
0000 1001
0000 1010
0000 1011
0000 1100
0000 1101
0000 1110
0000 1111
1111 1111
```

The problem seems to be that the data is not properly loaded before the first clock edge, which is a problem if there are other circuits dependent on that value. 

I am attaching the circ and mem files.

[ram-test.zip](https://github.com/logisim-evolution/logisim-evolution/files/10493892/ram-test.zip)

maehne, 2023-01-27:
@ChristopherPAndrews: Thanks for providing a test case for this issue!

mbaillif, 2023-08-14:
I don't know if it's the same problem but 
I'm testing this simple circuit ROM
[image]
And if I run the following commands with two versions of logisim (3.7.0 and 3.8.0) 
It is observed that version 3.8.0 does not work.

java -jar logisim-evolution-3.7.0-all.jar memory_test.circ -tty table -load mem1.txt 
Adr0   Out0
0000 0x00
0001 0x55
0010 0x9f
0011 0xbd
0100 0xcf
0101 0xd8
0110 0x6b
0111 0x9c
1000 0xb7
1001 0x78
1010 0x47
1011 0xd2
1100 0xb7
1101 0x2c
1110 0xa2
1111 0x8a

java -jar logisim-evolution-3.8.0-all.jar memory_test.circ -tty table -load mem1.txt
Adr0   Out0
0000 0x00
0001 0x00
0010 0x00
0011 0x00
0100 0x00
0101 0x00
0110 0x00
0111 0x00
1000 0x00
1001 0x00
1010 0x00
1011 0x00
1100 0x00
1101 0x00
1110 0x00
1111 0x00

maehne, 2023-08-14:
@mbaillif: Could you please test whether this gets fixed by PR #1824 for issue #1747?

mbaillif, 2023-08-16:
I'm sorry, but my example is a stupid mistake on my part. 
1- Only RAM is considered with the -tty table -load option.
2- This only works in simulation mode with a clock and a halt signal.

However, it shows that version 3.9.9dev does not initialize data randomly when the circuit is loaded. if the option is selected.

commit 1bacb21d (2023-08-19): Merge pull request #1830 from figurantpp/fix-initial-zero-ram-error

TTY now loads RAM before first propagation - Fixes #1338
closed 2023-08-19 by BFH-ktt1