logisim-evolution/logisim-evolution #1339: Canceling a wiring operation with right click can result in wrong starting point for next wire

flarn2006, 2021-12-25:
If I start drawing a wire, I noticed I can cancel it by right-clicking while drawing, like in MS Paint and some other programs. However, if I then start dragging from a different point to make a wire, it'll instead start the wire from the previous point.

https://user-images.githubusercontent.com/687313/147390204-189bf42c-1afa-4931-bbc2-180df7c9c0b0.mp4

This issue can be worked around by releasing the mouse button to place the affected wire, then undoing and starting again from the correct point.

maehne, 2021-12-27:
Thanks @flarn2006 for the bug report! I can confirm this confusing behaviour.

dbh201, 2022-01-15:
fixed in pull request

BFH-ktt1, 2022-03-08:
Can this issue be closed?

maehne, 2022-03-13:
@BFH-ktt1: From my side yes.

@flarn2006: Do you agree as original poster of this issue?

closed 2022-03-16 by BFH-ktt1