logisim-evolution/logisim-evolution #1347: Cannot add VHDL Entity into an old project

SonoSooS, 2022-01-01:
I have an old project I've been working on for quite some time, but it became a mess to maintain, so I decided to learn VHDL.
Luckily for me, this beautiful project has VHDL support!

However, trying to add VHDL support to my existing project, I get the dreaded "The simulator has crashed, save your work and restart Logisim" messagebox.

This is sadly not just an ignorable fluke, the simulation does bug out quite heavily.

VHDL support seems to work in a brand new project however, so this seems to be related to the old project file.

I would've attached log files if I could find any, but I can't find it anywhere I searched, and it's not described anywhere where to look for log files either, so I can't attach a stack trace sadly.

BFH-ktt1, 2022-03-16:
@SonoSooS : Could you please try if with my latest PR this issue has been fixed?

SonoSooS, 2022-03-16:
Doing some minimal testing, it seems to be working now! Adding a VHDL entity still glitches out the simulation a bit, but I don't think it's relevant to this issue. No crashes yet, and that's what matters.

I'll do some more vigorous testing, and report back if the it has been truly fixed.

Thanks a bunch for your effort to fix it! 🎉

SonoSooS, 2022-03-22:
Yes, this has indeed been fixed! Now it works the same as in a brand new project.

closed 2022-03-22 by SonoSooS