logisim-evolution/logisim-evolution #1350: VHDL subcircuit instantiation in VHDL is not supported

SonoSooS, 2022-01-02:
I have two VHDL circuits, a simple and a complex one. I want to include the simple circuit in the complex circuit.
I can't accomplish this due two two main reasons:
- The simple circuit is renamed to LogisimVhdlSimComp_1, and the complex one is renamed to LogisimVhdlSimComp_0
- The simple circuit has to be placed into the simulation for it to be compiled, but otherwise it's only used inside the complex circuit, and goes unused in the simulation (not connected)

These are the only reasons preventing subcircuits from working.

This is not working:
```VHDL
-- ENTITY declaration of complex circuit goes here

ARCHITECTURE TypeArchitecture OF circuit_complex IS
    COMPONENT circuit_simple IS
    PORT
    (
        -- Port map of simple circuit
    );
    END COMPONENT;
BEGIN
    instance_simple: circuit_simple
        PORT MAP
        (
            -- Port binding of simple circuit
        );
    
    -- Rest of the complex circuit code goes here
END TypeArchitecture;
```

This is working as expected:
```VHDL
-- ENTITY declaration of complex circuit goes here

ARCHITECTURE TypeArchitecture OF circuit_complex IS
    COMPONENT LogisimVhdlSimComp_1 IS
    PORT
    (
        -- Port map of simple circuit
    );
    END COMPONENT;
BEGIN
    instance_simple: LogisimVhdlSimComp_1
        PORT MAP
        (
            -- Port binding of simple circuit
        );
    
    -- Rest of the complex circuit code goes here
END TypeArchitecture;
```

The only thing changed in the code is that I'm using the renamed name instead of the in-project name to make it work.

sheyifan, 2025-02-25:
@SonoSooS 
I think this issue was caused by this line of [code](https://github.com/logisim-evolution/logisim-evolution/blob/8895e463ac998894443edd068a11eff8b576ee1d/src/main/java/com/cburch/logisim/vhdl/sim/VhdlSimulatorTop.java#L207). You can try to substitue the `label` with `((VhdlEntity)fact).getContent().getName()`. I have tried to modify this line so that I can successfully simulate VHDL subciruit in logisim.
Otherwise, I found that some of the modelsim version can not work correctly in logisim. Even if you rebuilt logisim, you can not make it right. ModelSim 20.1.0 works after applied the above patch, but other versions are not ensured to work.

hewzhew, 2026-08-20:
I traced this through the current VHDL simulation file-generation path and the 2019 change that introduced `LogisimVhdlSimComp_<n>`.

The suggested one-line substitution is close to the naming problem, but using the source entity name as the simulation name would not be safe by itself:

- `VhdlSimulatorVhdlTop` currently uses the simulation name both as the VHDL component type and as the unique prefix for top-level ports and instance labels.
- Two unlabelled instances of the same VHDL entity would therefore generate duplicate names such as `circuit_simple_input` and `circuit_simple_map`.
- If an instance has a user label, `VhdlEntity.setSimName()` currently ignores the supplied fallback name and derives another renamed entity name from that label, so the one-line change would not consistently preserve source-level references.
- This uniqueness was introduced deliberately in commit `5c92076fd` to fix simulation of multiple/unlabelled VHDL instances.

I think the robust boundary is to separate two concepts which are currently represented by one string:

1. **Canonical entity name** — the name declared in `VhdlContent`, used for the VHDL source file, component type, and references between project-local VHDL entities.
2. **Unique bridge/instance name** — the existing `LogisimVhdlSimComp_<n>` (or another unique internal name), used only for generated top-level signal names, instance labels, and the Logisim/Tcl socket bridge.

For new-style project VHDL entities, the generated simulation top could then declare each canonical component type once, instantiate it one or more times using unique bridge names, and keep the Tcl paths unique. The old-style per-instance `VhdlEntityComponent` path can remain unchanged initially.

The second part of the report is dependency discovery. `VhdlSimConstants.getVhdlComponents()` only discovers placed instances, while `LogisimFile.getVhdlContents()` already exposes all project VHDL definitions. A bounded first implementation could compile the validated project-local VHDL definitions under their canonical names, including definitions which are not placed on the canvas. That would make the report's component-declaration example resolvable without adding an unused Logisim instance.

I would add generation-level regression tests that do not require Questa/ModelSim:

- one placed complex entity plus an unplaced project-local dependency;
- two instances of the same entity produce one canonical component declaration and two unique maps/port prefixes;
- labelled and unlabelled instances remain collision-free;
- generated source and compile scripts retain the canonical dependency names.

Would compiling all validated project-local VHDL definitions be an acceptable first scope, or would you prefer dependency scanning (and an explicit supported set of component/direct-entity instantiation forms) from the start?

maehne, 2026-08-21:
IMHO, all VHDL entities added to the user's library should get analysed so that they can be resolved during elaboration by the simulation or synthesis tool. Unfortunately, this does not liberate us from doing dependency scanning, as an entity and its architecture(s) needs to have been analysed before it can be instantiated directly into another architecture. Only when using indirect instantiation via a component declaration, the binding of the actual design entity (i.e., the entity(architecture) pair) gets deferred to elaboration time.

There is a way to side step the dependency scanning: Logisim-evolution allows customising the order in which circuits are listed in the library pane via the small up and down arrow icons on the top. By doing then the analysis of the VHDL entities in that order, we can shift the dependency resolution on the shoulder of the user. If analysis yields an instantiation error, a pop up window could remind the user to check the order of circuits in his library to ensure dependencies are met.

Personally, I recommend to my students to always instantiate entities with the wanted architecture explicitly instead of via a component unless the user wants to exploit configurations to simulate different implementation variants with the same test bench or he needs to instantiate Verilog or Spice representations of his sub-circuits.

That said, your analysis is right: For instantiation, the entity name needs to be used, but a (at this scope) unique instantiation name needs to be derived from it. Personally, I would prefer if that name allows to directly link it to the declared entity instead of using a generic `LogisimVhdlSimComp_<n>` name.

Please note that the VHDL co-simulation part of Logisim-evolution is fragile and did not see much maintenance, as the primary maintainers are not actively using it. There are many other related issues to be resolved to make it more robust and support also open-source simulators such as ghdl or nvc (see #84).