logisim-evolution/logisim-evolution #1360: defect: use of Demux component, stops synthesis

primechaser, 2022-01-11:
Hi Team,

there is an issue when using Demux component in Logisim 3.7.2.

attached example causes synthesis to stop due to the following error (taken from Quartus II after loading the project)

"Error (10482): VHDL error at Demultiplexer_4_behavior.vhd(16): object "enable" is used but not declared"

thanks for looking into this.

[demux_issue.zip](https://github.com/logisim-evolution/logisim-evolution/files/7848684/demux_issue.zip)

jeritt, 2022-01-15:
I also noticed this behavior. I had a similar problem while trying to output in Verilog and Xilinx tools. Upon some digging, it appears that even when the demux is not using an enable pin in the Logisim Evolution symbol, the variable for enable is created but missing a declaration or other references in the .V file.

BFH-ktt1, 2022-03-08:
fixed in my latest push, closing

closed 2022-03-08 by BFH-ktt1