logisim-evolution/logisim-evolution #1363: defect: RAM component sets wrong address size in HDL code

primechaser, 2022-01-12:
Hi team, 

after correcting defect #1358 manually in generated HDL code, I realized another issue.

attached example holds 2 RAM components

a) schedule_RAM
b) hash_RAM

where schedule RAM is sized 64 x 32 Bytes, Hash RAM is using 8 x 32 Bytes.
the generated HDL code for hash RAM shows the following entry....

SIGNAL s_addressReg    : std_logic_vector( **5 DOWNTO 0** );

this setting is wrong, as it should show
SIGNAL s_addressReg    : std_logic_vector( **2 DOWNTO 0** );

additionally you might also want to correct the following entry:

   TYPE memoryArray IS ARRAY ( 8 DOWNTO 0 ) OF std_logic_vector( **31DOWNTO 0**);
(missing space)

Thanks for looking into this

[RAM issue.zip](https://github.com/logisim-evolution/logisim-evolution/files/7852339/RAM.issue.zip)

primechaser, 2022-03-16:
thanks for having this fixed. I had a quick test and can cofirm vector range looks fine now.
still I see the array size issue has not been addressed yet

primechaser, 2022-03-16:
thanks for having this fixed. I had a quick test and can confirm issues above look good now

do you want me to close this defect?

BFH-ktt1, 2022-03-16:
Closing issue

closed 2022-03-16 by BFH-ktt1