logisim-evolution/logisim-evolution #1364: Output register value is incorrect after 4 bits, this is a bug

RepolhoIguinorante, 2022-01-13:
The register has a bug in its output, when connected by a wire to a BCD for Hexadecimal display, when the number exceeds 4 bits, it starts to generate some random numbers...

MarcinOrlowski, 2022-01-13:
Please include project file (as ZIP) demonstrating the issue. What version of LSe it is?

RepolhoIguinorante, 2022-01-13:
Ok

RepolhoIguinorante, 2022-01-13:
I already posted the example project...

RepolhoIguinorante, 2022-01-13:
> Please include project file (as ZIP) demonstrating the issue. What version of LSe it is? 

My version of LogiSim Evolution is 3.7.2 is the latest

RepolhoIguinorante, 2022-01-14:
[image]

RepolhoIguinorante, 2022-01-14:
You got it?

In the register we have a number, but in the wire we have another...

This bug gives BCD confusion. The Hexadecimal Display for example, shows a totally wrong number of what should be shown...

RepolhoIguinorante, 2022-01-14:
Please fix this...

BFH-ktt1, 2022-01-14:
I'm unsure what you mean. From your picture the value stored in the register is 0xaa, which correspond to 10101010b shown on the wire, which in respect corresponds to -86 when using 2's Complement interpretation. Hence all shown in the picture is correct. Please be more specific on the error you think is present.

RepolhoIguinorante, 2022-01-14:
Thanks for the reply...But, (0×aa) or (aa) is equal to 170 and not to -86...

BFH-ktt1, 2022-01-14:
10101010b equals 170 in "unsigned" binary interpretation, 10101010b equals -42 in sign and magnitude "signed" interpretation, and 10101010b equals to -86 in two's complement "signed" interpretation. Hence it depends how you interpret this set of bits. Therefore I do not see a bug here, as the numbers are correct.

BFH-ktt1, 2022-01-14:
[image]
You can change the interpretation in the settings under ```Secon radix when wire poked```, which is default on ```Signed Decimal```(2's complement). When you change it to ```Unsigned Decimal``` the 170 will appear. Closing issue as I do not see a bug here.

closed 2022-01-14 by BFH-ktt1