logisim-evolution/logisim-evolution #1371: User Experience

Claupio, 2022-01-16:
I have problems in circuit selection (explorer panel); when I double click to a circuit it does't "appears" or be selected; when I modify the appearence of a circuit the modifies were not saved

Claupio, 2022-01-16:
P. S. I use this software from windows

Claupio, 2022-01-16:
P.P.S. For the circuit selection: the "click event" is in the icon but not in the label

dbh201, 2022-01-16:
To set a custom circuit appearance:
Select the circuit in the schematic or explorer panel, go to the properties tab at the mid-left, and set the `Appearance` attribute to `Custom`. That will change the appearance of that circuit type.

When you select `Edit Circuit Appearance` again, are your changes saved there?

Claupio, 2022-01-16:
Yes, thanks