logisim-evolution/logisim-evolution #1376: defect: "generic" entry is not getting copied into HDL

primechaser, 2022-01-20:
Hi team,

attached example shows an HDL entity with the following entry:

ENTITY generic_entry IS
  **Generic ( clk_freq		: integer   :=   128       	-- frequency of input clk in Hz
             );** 
  PORT (clock      : IN  std_logic;....

the synthesized code doesn't hold the "Genric" statement anymore

Error (10482): VHDL error at generic_entry_behavior.vhd(26): object "clk_freq" is used but not declared

the generic _entry_entity.vhd file starts with:	

ENTITY generic_entry IS
   PORT ( clock : IN  std_logic; .....

thanks for looking into this

[generic entry missing.zip](https://github.com/logisim-evolution/logisim-evolution/files/7905399/generic.entry.missing.zip)

BFH-ktt1, 2022-01-23:
This is correct, as generics are not supported by the current VHDL-implementation in logisim. Currently there is no effort going on on implementing generic support in logisim. PR's are welcomed.

hewzhew, 2026-08-20:
I rechecked this against current `main`. Generic support is now partially present: `VhdlParser` and `VhdlContent` parse `integer`, `natural`, and `positive` generics and retain their defaults, while `VhdlEntityAttributes` exposes per-instance overrides. The synthesis omission remains because the generic hooks retained in `VhdlHdlGeneratorFactory` are commented out and predate the current HDL generator API.

I propose a bounded, behavior-preserving synthesis slice:

- add a protected VHDL parameter-declaration hook to `AbstractHdlGeneratorFactory`, defaulting to its existing `HdlParameters` behavior;
- override it for VHDL entities so generated entity declarations preserve each parsed generic's name and exact type;
- override the existing `getParameterMap(AttributeSet)` hook so each instance emits its configured value or the parsed default in its `generic map`;
- leave VHDL editing, simulation, ports, libraries, and architecture text unchanged.

Focused tests can verify generated declarations and both default and overridden instance values without requiring an external synthesis tool. Does this scope match the intended direction for #1376?

maehne, 2026-08-21:
@hewzhew: The outlined approach sounds good to me.