logisim-evolution/logisim-evolution #1377: defect: RAM component shows missing clock connections

primechaser, 2022-01-21:
Hi team, 
attached example shows a RAM component, which fails on my DE0 board (after open issues have been fixed manually) due to unconnected clock signal.

For comparison reasons I added a register to understand which clock lines logisim usually hooks on to

here is an excerpt of main_behaviour.vhd

   --------------------------------------------------------------------------------
   test_RAM : RAMCONTENTS_test_RAM
      PORT MAP ( address => s_logisimBus9(3 DOWNTO 0),
                 **clock   => '0',**
                 dataIn  => s_logisimBus2(3 DOWNTO 0),
                 dataOut => s_logisimBus0(3 DOWNTO 0),
                 oe      => s_logisimNet14,
                 **tick    => '0',**
                 we      => s_logisimNet8 );

we see "clock" and "tick" is not connected

so after correction it might look like the following

   --------------------------------------------------------------------------------
   test_RAM : RAMCONTENTS_test_RAM
      PORT MAP ( address => s_logisimBus9(3 DOWNTO 0),
                 clock   => logisimClockTree0(4),
                 dataIn  => s_logisimBus2(3 DOWNTO 0),
                 dataOut => s_logisimBus0(3 DOWNTO 0),
                 oe      => s_logisimNet14,
                 tick    => logisimClockTree0(2),
                 we      => s_logisimNet8 );

other circuits assign '1' to "tick", so not sure what drives this....

additionally my understanding is the array size is one bit too large:

TYPE memoryArray IS ARRAY ( 16 DOWNTO 0 ) OF std_logic_vector( 3 DOWNTO 0);

shouldn't it be 15 downto 0 in this case?

thanks for looking into this

[RAM missing clock.vhd.zip](https://github.com/logisim-evolution/logisim-evolution/files/7910925/RAM.missing.clock.vhd.zip)

primechaser, 2022-03-16:
thanks for having this fixed. I had a quick test and can cofirm clock connections look fine now.
still I see the array size issue has not been addressed yet

BFH-ktt1, 2022-03-16:
Closing issue

closed 2022-03-16 by BFH-ktt1