logisim-evolution/logisim-evolution #1378: Buffers holding state

Phosphorjr, 2022-01-23:
[image]
pictured here, there is no input to the buffer, but it stays at a 1
the simulation is not paused

BFH-ktt1, 2022-01-23:
I cannot reproduce this behavior, could you send a zipped circ file that reproduces this behavior?

Phosphorjr, 2022-01-23:
[Repeater.zip](https://github.com/logisim-evolution/logisim-evolution/files/7920593/Repeater.zip)
this Repeater circuit here
its function is, on signal input, repeat oscillating on/off for the specified number of times in the counter, then stop at 0, all within one clock cycle
the circuit works sometimes, but only sometimes

BFH-ktt1, 2022-01-25:
Such kind of circuits will never correctly work in logisim, as the simulation engine of logisim is based on the "zero-delay" model (Although gates have "some" delay to prevent dead-locks). Hence if you want to build asynchronous systems that relay on gate-delays you have to use another tool, as making logisim simulate correctly the gate-delays would require a complete rewrite of the logisim engine (the reason why Carl Burch started a new project).
Maybe you could try digital for this purpose.
Hope this explains your issue.

Phosphorjr, 2022-02-01:
this does explain my issue, thank you.

closed 2022-02-01 by Phosphorjr