logisim-evolution/logisim-evolution #1379: Custom appearance does not always get saved

maehne, 2022-01-23:
To reproduce:

1. Add a new circuit
2. Place some input / output pins
3. Edit custom appearance: Only resize the box and shift the pins
4. Set appearance of circuit to "Custom"
5. Place the circuit: It should have the new appearance
6. Save the circuit file
7. Close the circuit file
8. Reopen the circuit file: The placed circuit has the default custom appearance without resized box and shifted pins

This was reported by @Claupio in discussion #1373.

Further: I would suggest that the circuit appearance attribute is set automatically to "Custom" when the user modifies the circuit appearance. I think that is the expected intuitive behaviour. To restore the default appearance, the user shall switch the appearance to one of the default appearances, e.g., "Logisim-evolution". Users may also appreciate a "Reset custom appearance" function as a starting point for an overhaul of the custom appearance.

taiji1985, 2022-01-29:
I got the same problem .

Lars-Private, 2022-03-23:
[image]
You have to click on your custom circuit and choose your custom appearance

maehne, 2022-03-28:
@Lars-Private: Thanks for your comment, but you seem to have misread the description of this issue. I am aware that you have to manually set appearance to "Custom" (and I suggest that it is set to this value automatically upon editing the appearance). However, the main issue is that in some cases the modified appearance does not get saved to the `.circ` file.

carlini, 2022-04-30:
I've figured out the root cause but not sure the best way to fix it. 

The reason this bug occurs is this line returns false

https://github.com/logisim-evolution/logisim-evolution/blob/49398db3f0e90793901ba2a1581c982bbdcdf31a/src/main/java/com/cburch/logisim/file/XmlWriter.java#L273

even if you've modified the appearance of a circuit, which is wrong. The actual bug occurs here

https://github.com/logisim-evolution/logisim-evolution/blob/49398db3f0e90793901ba2a1581c982bbdcdf31a/src/main/java/com/cburch/logisim/circuit/appear/CircuitAppearance.java#L102

which returns true even if the object has been modified by moving or resizing. The reason this happens is because when the default appearance is created it's also copied into the current appearance

https://github.com/logisim-evolution/logisim-evolution/blob/49398db3f0e90793901ba2a1581c982bbdcdf31a/src/main/java/com/cburch/logisim/circuit/appear/CircuitAppearance.java#L75-L76

and so the same objects are shared in both lists, so when the user modifies the visual appearance it also modifies the "default" appearance so the test returns false.

A hack to fix this is to re-run the line 

```java
defaultCustomAppearance = DefaultCustomAppearance.build(circuitPins.getPins()); 
```

but I'm not sure if that's the best solution.

A workaround in the meantime is to just add any other object to the appearance (e.g., some text, a line, another box) and then the condition here will always cause the test to return `false`:

https://github.com/logisim-evolution/logisim-evolution/blob/49398db3f0e90793901ba2a1581c982bbdcdf31a/src/main/java/com/cburch/logisim/circuit/appear/CircuitAppearance.java#L94

maehne, 2022-05-02:
Thanks for the analysis @carlini! It should help @BFH-ktt1 or someone else to propose a PR to fix it.

dbh201, 2022-07-28:
https://github.com/logisim-evolution/logisim-evolution/blob/49398db3f0e90793901ba2a1581c982bbdcdf31a/src/main/java/com/cburch/logisim/circuit/appear/CircuitAppearance.java#L471

I think this line was intended to deep-copy the `shapesBase` list, which would have fixed this bug. Unfortunately, there is no convenient way (that I know of) in Java to deep-copy lists or other collections of non-primitive objects. You have to manually iterate over the objects and call clone() for each one; since `CanvasObject` implements `Cloneable`, this should work.

Thanks to @carlini for finding the issue. I'll put it in the PR.

BFH-ktt1, 2022-08-26:
This is fixed by #1380. Closing. Please reopen if problem persists

closed 2022-08-26 by BFH-ktt1