logisim-evolution/logisim-evolution #1385: 7447 behavior error

quick0831, 2022-01-30:
I was trying to use 7447 for my circuit, but I found that the truth table of the IC is incorrect. It act more like a 7448.

One is that the output is negated (like 7448 do).

The other is that for input value over 9, it does not produce the correct output according to the datasheet. Instead it displays A-F.

[Datasheet for 7447 & 7448](https://www.ti.com/lit/ds/symlink/sn74ls47.pdf)

quick0831, 2022-01-30:
BTW forgot to mention, I'm using v3.7.2

maehne, 2022-02-02:
@MarcinOrlowski: Would you maybe willing to have a look?

quick0831, 2022-02-03:
I also used 74139 in my circuit, and I experience the same issue as stated in #1334 (kind of not important because investigated)

quick0831, 2022-02-03:
I'm trying to look into the code

quick0831, 2022-02-03:
Tried to build from develop branch

[image]

quick0831, 2022-02-03:
(the output is still inverted)

quick0831, 2022-02-03:
I attempted to invert the output of the code, and run a small test with the datasheet

[image]

quick0831, 2022-02-03:
(the 7 segment display is set to active low)

BFH-ktt1, 2022-03-08:
Merged, closing issue

PR #1388 (2022-03-08): Invert the output of the DisplayDecoder
Edit the output value of the decoder to match the datasheet

Fix #1385

[Datasheet](https://www.ti.com/lit/ds/symlink/sn74ls47.pdf)

* #1385
closed 2022-03-08 by BFH-ktt1