logisim-evolution/logisim-evolution #1386: Transistor not functioning correctly.

Digital-Johnson, 2022-01-31:
The N-channel transistor in Logisim Evolution v3.7.2 does not function correctly. I have included a test circut below.
It is a simple N-channel buffer circut. When the input goes high so should the output. However, there is no change to the output. The P-channel transister works as expected.

[Transistor test.zip](https://github.com/logisim-evolution/logisim-evolution/files/7967484/Transistor.test.zip)

davidhutchens, 2022-01-31:
Please read #1295. This appears to be a similar issue.

BFH-ktt1, 2022-01-31:
The rationale behind this label is that logisim's simulation engine is a digital one. Providing the correct behavior of the transistors (FET's) would require an analog simulation engine (like Spice/LTspice, etc.) which is out of the scope of the goals of logisim. The simulation of the transistors in logisim is only the off-state and the saturation state in case the p-channels are connected to VCC and the n-channels are connected directly to GND, which makes sense for digital circuits (hence push-pull configurations, emittor/drain followers are not supported, as the circuit you provided).

closed 2022-02-02 by maehne