logisim-evolution/logisim-evolution #1390: Slow / Not working Synthesize

EmperorPenguin18, 2022-02-21:
If I try to synthesize my circuit, it hangs on certain sub-circuits. Logisim is using a substantial amount of CPU during this time, so I left it running for a couple days but had no progress. If I remove these sub-circuits it works. The settings I'm using are TERASIC DE0 and main as the top-level. I've attached the circuit file.

[cpu_phase1.txt](https://github.com/logisim-evolution/logisim-evolution/files/8112208/cpu_phase1.txt)

primechaser, 2022-02-22:
you might want to look into #1309, as you are using bitextenders extensively in your circuits. I believe, the logisim team will apply the fix in the upcoming release. for the time being you could replace your bitextenders with splitters

EmperorPenguin18, 2022-02-23:
Using the develop branch fixes the issue. Thanks for the help. Is there an ETA for the next release?

closed 2022-02-23 by EmperorPenguin18

MarcinOrlowski, 2022-02-23:
We do not have fixed roadmap, but you can always use our nightly builds if you are looking for binaries built from recent code.

See https://github.com/logisim-evolution/logisim-evolution#nightly-builds for more info