logisim-evolution/logisim-evolution #1394: Behavior of XOR-Gate

jbtls, 2022-02-25:
If you set the Number of Inputs higher than 2 the resulting truth-table looks like this:
```
c b a | x
~~~~~~~~~
0 0 0 | 0
0 0 1 | 1
0 1 0 | 1
0 1 1 | 0
1 0 0 | 1
1 0 1 | 0
1 1 0 | 0
1 1 1 | 0
```
Wikipedia and some of my colleagues, on the other hand, say that: 
> XOR gate (sometimes EOR, or EXOR and pronounced as Exclusive OR) is a digital [logic gate](https://en.wikipedia.org/wiki/Logic_gate) that gives a true (1 or HIGH) output when the number of true inputs is odd [(https://en.wikipedia.org/wiki/XOR_gate](https://en.wikipedia.org/wiki/XOR_gate)
If this definition is correct, the resulting table should look like that:
```
c b a | x
~~~~~~~~~
0 0 0 | 0
0 0 1 | 1
0 1 0 | 1
0 1 1 | 0
1 0 0 | 1
1 0 1 | 0
1 1 0 | 0
1 1 1 | 1
```
Hope together we could solve this issue...

jbtls, 2022-02-25:
Have also found this: https://en.wikipedia.org/wiki/XOR_gate#More_than_two_inputs

BFH-ktt1, 2022-02-25:
Basically there does not exists one definition of a XOR-gate with more than 2 inputs, it can be either a odd number of inputs or a one-hot detection. Therefore I always suggest to take only XOR-gates with 2 inputs and cascade them. In this case you get a circuit that detects an odd number of "1"'s

jbtls, 2022-02-25:
Thank you for the quick feedback. This is also the solution we came up with. In this case I would recommend either to adjust the behavior of the XOR gate as described above or to program it in a way that you can not set more than two inputs on one xor gate.

BFH-ktt1, 2022-02-25:
You can select the behavior of a XOR-gate with more than 2 inputs with it's attributes, hence you can select the behavior you require.

jbtls, 2022-02-25:
Oh, thank you! I had overlooked that. Thanks for the great work on the project!

closed 2022-02-25 by jbtls