logisim-evolution/logisim-evolution #1396: Counter resets althought wire is not active

MaxiDoet, 2022-03-01:
I'd like to build a clock divider but my counter gets reset by the comparator althought the values changed and the wire is not active

[image]

Here is my project:
If you test it don't forget to set the divider
[Project.zip](https://github.com/logisim-evolution/logisim-evolution/files/8161495/Project.zip)
[image]

davidhutchens, 2022-03-01:
The circuit behavior seems to work for me. The reset input is asynchronous, so it resets as soon as the value of the counter hits the div value and does not wait for the next clock.

I suspect the behavior you desire can be obtained by moving the output of your OR from the R input to the M1 [load] input. Then it will wait for the next clock cycle to load the zero you have as the data input to your counter.

Does this solve the problem?

MaxiDoet, 2022-03-01:
Have you opened the circuit and tried it? The counter resets even if no line is active thats my problem

davidhutchens, 2022-03-01:
Yes, I did open the circuit and try it. And I do not see it resetting when no line is active. For example, if you change the value of div to 4, it resets after counting to 3. When it hits 4, the equality test is true for a very short instant which resets the counter to 0 which causes the equality to drop back to false. This happens quickly, before the next clock. Thus you never "see" the value at 4 or the Reset line as true.

MaxiDoet, 2022-03-01:
Oops, my mistake. 
This is how i solved it: 
[image]

closed 2022-03-01 by MaxiDoet

maehne, 2022-03-01:
It is unreliable to use the asynchronous reset to implement the desired reset behaviour for the counter described by you. The reason are hazards due to the generation of the asynchronous reset signal with the help of a combinatoric circuit. Your fix above is fragile and might not work if the circuit gets synthesised on real hardware.

Instead, it is recommended to implement a synchronous reset behaviour. In the case of your circuit example, it should suffice to connect your reset condition signal to the load input. Loading zero into the counter at a rising clock edge corresponds to a synchronous reset.