logisim-evolution/logisim-evolution #1402: No enable with DFlipFlop

ArnaudMartinCaen, 2022-03-06:
Ii it possible to add a enable port with DFlipFlop, like logisim software.

Thank yo for your help.

BFH-ktt1, 2022-03-07:
If you need an enable, please use the register component

ArnaudMartinENSI, 2022-03-08:
Thank you for your help. But there is no "Set" Pin on the register component.
I need a A flp flop, with reset, set and enable.

Have a good day.

BFH-ktt1, 2022-03-08:
You can than "just" put a mux in front of the flipflop that either keeps the current state or takes the new state.

ArnaudMartinENSI, 2022-03-11:
Thank you for your answer and your help. I did that and it works, but my schematic is largest.

BFH-ktt1, 2022-03-11:
Closing this issue

closed 2022-03-11 by BFH-ktt1

maehne, 2022-03-11:
@ArnaudMartinCaen: You can put the D-FF with the Mux in front into a sub-circuit and draw a custom appearance, which resembles the D-FF and adds the requested enable pin. If you use this sub-circuit then in your circuit, it should become more compact. Does this address your need better?