logisim-evolution/logisim-evolution #1415: Research and perhaps move main method tests to junit

davidhutchens, 2022-03-15:
The code contains 4 main methods that may be intended as tests. These should be move to junit tests as appropriate. The four main methods are:

src/main/java/com/cburch/hex/Test.java:  public static void main(String[] args) {
src/main/java/com/cburch/logisim/util/AlphanumComparator.java:  public static void main(String[] args) {
src/main/java/com/cburch/logisim/gui/hex/HexFile.java:  public static void main(String[] args) {
src/main/java/com/cburch/logisim/analyze/gui/Analyzer.java:  public static void main(String[] args) throws Exception {

maehne, 2022-05-05:
@davidhutchens: Can this be closed since PRs #1438 and #1446 got merged in the meantime?

davidhutchens, 2022-05-05:
@maehne: PRs #1438 and #1446 moved the main test method from HexFile to junit. There are three other main methods listed in this issue that should be examined. So I will leave this issue open pending a resolution for those.

davidhutchens, 2024-02-06:
With the merge of #1955, all the main test drivers have been converted to JUnit.

closed 2024-02-06 by davidhutchens