logisim-evolution/logisim-evolution #1419: Flatpak install cannot flash to boards

SekoiaTree, 2022-03-22:
After several grueling hours of debugging and reading through the source code (thank you so incredibly much for being open-source) to know what commands to run, and never getting any errors when I was through logisim, I have identified my problem.

It was fucking flatpak.

Anyway, when trying to download a circuit to a board, I got a strange error in stage "Optimizing Altera project". The strange part was that there *was* no error. It simply immediately failed, with no message at all; it did the pretty "===> Optimizing Altera project" text and failed with no text after. It couldn't have been an IOException or InterruptedException, as the error message only matched up to the exit code not being 0. After installing through the deb instead of flatpak, it works perfectly. Still no idea what caused it, but the cause is basically certainly flatpak.

This issue is half as a warning, half as a google search hit for others searching so they don't have to go through what I did. Sorry if it's a bit rambly, my emotions are still running high from finally figuring it out.

maehne, 2022-04-01:
@SekoiaTree: Thanks for the report, but please be a bit more careful in the choice of you words next time! Some people might get upset. There might be more to this issue though. I had several occasions were students reported similar issues with Logisim-evolution's FPGA commander being stuck in the "Optimizing Altera project" phase. However, in those cases it seemed more related with the particularly poor quality of the schematics they tried to synthesize. So, this might deserve some further investigation. PRs trying to make this part of the code more robust are welcome!

SekoiaTree, 2022-04-01:
Fair enough, edited!

maehne, 2022-04-01:
Thanks @SekoiaTree!

SekoiaTree, 2022-04-01:
I don't think my schematic was too bad, though; potentially, it's simply the first step where any issues can really happen?

maehne, 2022-04-01:
I am not sure. However, it would help if we have a circuit, which triggers the issue. Could you maybe add the problematic `.circ` file in a ZIP file to a message and specify which sub-circuit you tried to synthesize?

SekoiaTree, 2022-04-02:
I tested it, any circuit fails, including simply 1 input connected to 1 output.