logisim-evolution/logisim-evolution #1420: Bug in Combational Analysis simplification function

BlackGoku36, 2022-03-24:
To minimize: (x+y) (~x+z) (y+z)
-> (y+x) (y+z) (~x+z)                   [Re-arrange]
-> (y+xz) (~x+z).                           [(A + B) (A + C) = A + BC]
-> ~xy + yz + x~xz + xzz
-> ~xy + yz + xz

We get: `~xy + yz + xz`
But the circuit made by combational analysis gives: `xy + yz + ~xz`

[image]

Tested on:
- MacOS Monterey 12.1
- Logis Evolution v3.7.2

Hope these are enough info

BFH-ktt1, 2022-03-24:
Sorry, I cannot reproduce this bug, and furthermore the simplification of this circuit is: $$\overline{x}\cdot y + x cdot z$$. could you provide the .circ file that generates this bug and a description how you come to the result (the steps).

BFH-ktt1, 2022-03-24:
If you use the maxterms you get:
[image]
Hence in both cases correct....

BFH-ktt1, 2022-03-25:
All the screen-shots you show above show the correct behavior of the analyzer, in the first screen it implements the circuit you typed in as expression, and in the second and third window it creates the minimal form using the sum-of-product or the product-of-sum form. So where you think that there is a bug?

BlackGoku36, 2022-03-26:
Oh, I thought it simplifies by itself when you generate from typed expression, I got bit confused there. Thank you for your time.

closed 2022-03-26 by BlackGoku36