logisim-evolution/logisim-evolution #1421: Project won't synthesize with RAM

EmperorPenguin18, 2022-03-25:
I'm trying to synthesize my project with the Terasic DE0 setting and if there is a RAM component in the circuit, that circuit stops the process.

BFH-ktt1, 2022-03-26:
Please try the nightly build, this problem should have been resolved.

EmperorPenguin18, 2022-03-28:
Using the nightly build does not fix the issue.

maehne, 2022-03-28:
@EmperorPenguin18: Could you please provide us with a reproducer for the issue in form of a `.circ` file that exposes the issue? You'll need to put it into a ZIP archive to attach it to a message. Also, please specify which sub-circuit to synthesize and what message you are observing. More information about your platform (processor architecture, OS version, JDK version, Quartus Prime version, Logisim-evolution version) is also helpful to facilitate reproducing the problem on our side.

EmperorPenguin18, 2022-03-28:
[cpu_phase4.circ.zip](https://github.com/logisim-evolution/logisim-evolution/files/8366704/cpu_phase4.circ.zip)

It's the "memory" subcircuit. There's no message, the synthesization just hangs.
Architecture: x86_64
OS version: Linux 5.16
JDK version: openjdk-17
Quartus Prime version: N/A
Logisim-evolution version: develop/52ebf1bb (2022-03-24T08:39:14+0000)

BFH-ktt1, 2022-03-29:
@EmperorPenguin18 , thanks for your circuit, this helped me tracking down the bug. The problem lies in the "lineenables" that remove the OE-pin of the device. This causes an NPE when downloading. I will look into a bugfix asap.

EmperorPenguin18, 2022-03-31:
Is there a way to use the program with this fix? Would I have to compile it myself?

BFH-ktt1, 2022-03-31:
You can now use the nightly build that should have the issue fixed. Please reopen if the problem still persists

closed 2022-03-31 by BFH-ktt1