logisim-evolution/logisim-evolution #1424: Reload Library option not working

prlaba, 2022-03-31:
I have a large L-E project with many circuits and subcircuits.  I've organized my circuits into multiple .circ files that I can then load into libraries within other .circ files as needed.

If I modify one of my .circ files and then use the _Reload Library_ option to reload that library, nothing seems to happen.  None of the circuits in use from that library reflect the changes I made.

I confirmed that the Reload Library option works as expected in Logisim.

It's hard to believe this doesn't work in L-E, as there must be many users using this same option.

Am I missing something?  Are there instances where the Reload Library option would not reload the library without an error message or other indication?

maehne, 2022-04-01:
Thanks for reporting this issue! We are aware that using the `.circ` library feature has still some issues after all these years of development. In our courses, we recommend to our students to use instead the merge feature to selectively replace sub-circuit implementation in the monolithical `.circ` file containing all schematics of the whole system.

@BFH-ktt1: Could you have a look?

prlaba, 2022-04-01:
Thanks for looking into this!

I’m surprised to have hit this bug, given that the same _Reload Library_ option works correctly in Logisim.

Without a working _Reload Library_ option, it becomes extremely difficult to update a circuit when a change is made to its underlying circuit library (you have to delete any included circuits from that library, unload and load the library, then re-add the deleted circuits).

weszhuang, 2023-03-16:
> Thanks for reporting this issue! We are aware that using the `.circ` library feature has still some issues after all these years of development. In our courses, we recommend to our students to use [...]
The merge feature seems to lack any documentation on its function within the application.