logisim-evolution/logisim-evolution #1425: Shifters in Verilog

EmperorPenguin18, 2022-03-31:
When I synthesize to verilog, Shifter_2_bit.v, Shifter_32_bit.v and Shifter_64_bit.v all are missing a semicolon in the last assign statement.

maehne, 2022-04-01:
Thanks for reporting! The HDL code generation underwent last year a substantial reimplementation and, unfortunately, introduced some smaller bugs. On the VHDL side, these were found relatively quickly, but Verilog is used less on our side.

BFH-ktt1, 2022-04-04:
@EmperorPenguin18 , I confirm this bug, and also found another one where the stage results had the wrong name.

BFH-ktt1, 2022-04-05:
Please try the nightly build to see if the problem is fixed. Closing for now, please reopen when the problem persists.

closed 2022-04-05 by BFH-ktt1