logisim-evolution/logisim-evolution #1427: Support DE0-CV

EmperorPenguin18, 2022-04-01:
My school uses the DE0 and the DE0-CV interchangeably, so one day I may need to use one and another day need to use another. Logisim-evolution supports the DE0, and supports other Cyclone V based FPGAs, but not specifically the DE0-CV.

maehne, 2022-04-02:
We don't have access to such a board to create and *test* the FPGA board definition. However, you can create a configuration yourself using the built-in FPGA Board Editor. All you need is a suitable photo of the board and its reference documentation to identify the FPGA pins connected to the different usable I/O components on the development board. After loading the board photo into the editor, you basically draw the red squares over the components you want to use, specify their type, and fill in the required information from the reference manual. At the end, you save that board configuration to an XML file. In the FPGA Board Editor, you can also load the boards, which are already supported by Logisim-evolution to get an idea what information needs to be entered for the different components. Just double-click on the different red squares to see what information is associated with them.

If you provide us the XML file in a pull request, we can include it into a future Logisim-evolution version after review. The board definitions get stored to `src/main/resources/resources/logisim/boards/`.

EmperorPenguin18, 2022-04-04:
So I've created the XML file (thanks for the great instructions btw) and now I'm trying to test it with this circuit:
[de0_test.zip](https://github.com/logisim-evolution/logisim-evolution/files/8413419/de0_test.zip)

The problem is the synthesizer hangs at "Creating Download Scripts". I haven't seen this problem before so I don't know what it means.

Here is the board file:
[DE0-CV.zip](https://github.com/logisim-evolution/logisim-evolution/files/8413428/DE0-CV.zip)

BFH-ktt1, 2022-04-05:
There are some small issues in your board-description file:

1) at the FPGA-parameter you should specify `Package: FBGA 484` and `Speed grade: C7`
2) For the seven segment displays you cannot specify `null` for the decimal point. They must be connected to real pins of the FPGA. Maybe connect the decimal points to the address lines of the sdram, 

Hope this helps

BFH-ktt1, 2022-04-08:
Closing as merged by your PR

closed 2022-04-08 by BFH-ktt1