logisim-evolution/logisim-evolution #1437: Remove "Printer view" from Preferences->Layout

maehne, 2022-04-12:
We had so far 3 tickets (issues #862, #1324, #1434) plus several occasions in the classroom, where users thought simulation was broken in Logisim-evolution due they had inadvertently activated "Printer view" in Preferences-> Layout. So, it seems that this functionaly is causing more confusion than added value. I see its only value when exporting circuits as an image. IMHO, we should either remove the option "Printer view" from Preferences->Layout or add a very explicit warning about the undesired effect in the Layout tab or when activating the option.

davidhutchens, 2022-04-12:
I note that a user can print (or save) the image in "Printer view" by just selecting File/Print (or File/Export Image) from the menu. The only functionality that would be lost by removing the Preferences option is seeing that view on the screen before printing/exporting the image. I don't see a significant problem with removing the "Printer view" option from Preferences.

maehne, 2022-04-12:
Thanks for the feedback on this issue @davidhutchens!

BFH-ktt1, 2022-08-08:
Addressed by PR #1486. Closing. Please reopen if problem persists.

PR #1486 (2022-08-06): Remove "Printer view" from Preferences->Layout #1437
Fixes #1437

* #1437.
closed 2022-08-08 by BFH-ktt1