logisim-evolution/logisim-evolution #1439: 74715 has an inverted CLR

alexandref75, 2022-04-13:
74175 has a negative CLR but it is positive on the code:

    Fix 74175, CLR is inverted

diff --git a/src/main/java/com/cburch/logisim/std/ttl/Ttl74175.java b/src/main/java/com/cburch/logisim/std/ttl/Ttl74175.java
index 3374a2e71..19c19b535 100644
--- a/src/main/java/com/cburch/logisim/std/ttl/Ttl74175.java
+++ b/src/main/java/com/cburch/logisim/std/ttl/Ttl74175.java
@@ -52,7 +52,7 @@ public class Ttl74175 extends AbstractTtlGate {
       state.setData(data);
     }
     var triggered = data.updateClock(state.getPortValue(7));
-    if (state.getPortValue(0) == Value.TRUE) {
+    if (state.getPortValue(0) == Value.FALSE) {
       data.setValue(Value.createKnown(data.getWidth(), 0));
     } else if (triggered) {
       Value[] vals = data.getValue().getAll();

maehne, 2022-06-15:
@alexandref75: Thanks for the report. Could you please provide a PR against the `develop` branch?

BFH-ktt1, 2022-10-02:
Closing, as PR has been merged.

closed 2022-10-02 by BFH-ktt1