logisim-evolution/logisim-evolution #1441: Disconnected input to subcircuit changes to 0

prlaba, 2022-04-15:
Assume a master circuit with an input connected via a wire to the input of a subcircuit.

As expected, the subcircuit's input assumes the 0/1 value of the wire from the master circuit.

But if I disconnect the wire from the master circuit's input to the subcircuit's input, the subcircuit's input correctly shows U, but the wire leading from that input (and anything else connected to that wire), shows 0.  It should show U.

I tested this on Logisim and it gave the correct results (the subcircuit's wire connected to the U input also showed U).

maehne, 2022-04-18:
Thanks for reporting this issue! I can confirm the described behaviour. The event doesn't seem to get propagated properly. We are aware that the simulation kernel has some limitations and issues yielding it to end up in an inconsistent state. Unfortunately, we are not very familiar with its implementation.

prlaba, 2022-04-18:
Thanks.  In stepping through the propagation of signals during startup of my fairly complex circuit and subcircuits, I see other instances of this.  But this may be expected as a component or input feeds onto a wire (the wire’s input may be in a different state than the wire itself at a given stage of the propagation).