logisim-evolution/logisim-evolution #1442: logisim-evulution fails to run vsim albeit LM_LICENSE_FILE been set

diegodutraufrj, 2022-04-15:
I've set the LM_LICENSE_FILE to the correct file and also been able run vsim manually, however logisim-evultion fails with this message

Unable to checkout a license. Make sure your license file environment variable (e.g., LM_LICENSE_FILE)
is set correctly and then run 'lmutil lmdiag' to diagnose the problem.
Unable to checkout a license.  Vsim is closing.
** Error: Invalid license environment. Application closing.

diegodutraufrj, 2022-04-18:
Hi, @maehne sorry for the long delay I was unable to reboot my system to make sure the setup was configured as you've described.

I've tried both through the terminal and loading directly from the file explorer, in both cases I still got the same error, while I was able to use vsim directly (terminal and through Quartus). If I can provide any extra detail please let me know.

KInd regards,
Diego

maehne, 2022-05-05:
The cause is most probably that  the `vsim` launched in the background by Logisim-evolution is not seeing the necessary environment variable `LM_LICENSE_FILE`. Instead of a QuestaSim, we typically use ModelSim Altera Starter Edition, which does not require to pull a license from a license server. Therefore, we have not observed so far the issue. It would be good to know, what environment variable the `vsim` process sees. Maybe, you could try to create a wrapper (or edit temporarily the corresponding launch script) to dump the relevant environment variables to a file. 

Maybe, this [wiki page](https://help.ubuntu.com/community/EnvironmentVariables) can help you to find out, how to set the environment variable in a way that QuestaSim sees it upon launch from Logisim-evolution. It would be nice if you could report back your findings to help others, which may run into the same issue.

diegodutraufrj, 2022-05-06:
Hi @maehne, sorry for the delay. I prefer to edit the launch script by dumping the content of the LM_LICENSE_FILE in a file when I call it vsim directly the correct content is dumped, through logisim-evolution a garbage string "1650@eilic01" is dumped. I'll do another round of test to verify if setting this variable in a different place solves the problem or at least change the dumped data.

diegodutraufrj, 2022-05-06:
I've finished a few extra tests. I've put the LM_LICENSE_FILE in .profile and it doesn't work. Still push the same garbage out, however, when I call logisim through the terminal initially it does get the correct path however when I try to enable the VHDL simulation it gets the "1650@eilic01" string again.
I've also tried to use a ,desktop but w/o success

diegodutra, 2022-09-27:
Thanks @marsfan for find the issue in the code on #1548

diegodutraufrj, 2022-09-27:
Thanks @marsfan for find this issue in the code on #1548

BFH-ktt1, 2022-10-02:
Closing this issue due to #1548 and my PR #1551 that should fix the issue. Please reopen if it is not resolved.

closed 2022-10-02 by BFH-ktt1