logisim-evolution/logisim-evolution #1450: Component error

Fyn-D, 2022-05-03:
Hi. I've done a lot of work in Logisim evolution v2.13.8 using the transistor item. In this version the transistor worked perfectly, both N and P-type as shown in photo.
[image]

When I upgraded v3.7.2, the transistors no longer function as expected, as shown in photo
[image]

Please look into this as it means I'd have to modify all the work I've done changing all the transistor types and adding NOT gates.
This means several thousand alterations.
Thank you.

Also, the V3.x.x .jar files wont run on my Windows 10, but the v2 does. (My java run time is up-to-date)

maehne, 2022-05-05:
Indeed, the behaviour of the transistor component has been modified on purpose in Logisim-evolution to more accurately match their behaviour in typical digital gate circuit topologies (NMOS, PMOS, CMOS, pass transistors, pass gates). Still, this is only an approximation, as Logisim-evolution is a digital gate-level simulator and not an analogue transistor-level simulator (e.g., SPICE). For details, see similar [older issues](https://github.com/logisim-evolution/logisim-evolution/issues?q=is%3Aissue+transistor+is%3Aclosed) (e.g., issues #1386, #1295, #504, and #431).

Fyn-D, 2022-05-05:
Hi
Thank you for your swift response.

I understand the program is logic based.

But still, a transistor that works as a basic switch, in both N & P type,
is also used in basic logic circuits, not just analogue. I use BJT's mixed
in with 74 series logic chips all the time.

Could there not be an option in the properties to choose between (N, P, and
C MOS) and simple BJT style switching transistor ?

It would allow Logisim more scope.

Yours enquiringly
Simon

maehne, 2022-09-25:
@Fyn-D: Feel free to open PRs in this direction. However, make sure the new components including their usage restrictions are well documented. Though, be prepared that PRs can also get rejected after discussion if we feel that they don't fit with the overall philosophy of Logisim-evolution. Ourselves, we won't spend effort in this direction.