logisim-evolution/logisim-evolution #1452: Font size too large in the canvas

dxhisboy, 2022-05-04:
I'm not sure about if it was a system configuration problem on my computer. Most of my computers runs Archlinux with Java 18. It seems the pin text of RAM is too large and overlaps each other:
[image]

While I didn't found a way to configure font of the canvas in menus. I found that a modification in `InstancePainter` work:
```
  public InstancePainter(ComponentDrawContext context, InstanceComponent instance) {
    this.context = context;
    this.comp = instance;
    this.context.getGraphics().setFont(new Font("monospaced", Font.PLAIN, 12)); // added this line
  }
```
Is there a way to configure the font settings in a graphical way?

maehne, 2022-05-05:
Thanks for reporting the issue and proposing a fix! To understand better the cause of the issue, could you please answer the following questions?

- Does this problem occur by any chance on a high-resolution (4K) screen? 

- Do you have any zoom configured for the screen in your desktop environment? 

- Does it help to set the "zoom factor" to 1.0 in File -> Preferences... -> Window?

dxhisboy, 2022-05-06:
> * Does this problem occur by any chance on a high-resolution (4K) screen?

Screen: 4K, 32 inch (should not configure scale at that PPI)

> * Do you have any zoom configured for the screen in your desktop environment?

Desktop environment: GNOME 42 on Wayland, no scale configured.

> * Does it help to set the "zoom factor" to 1.0 in File -> Preferences... -> Window?

I've tried, it only scales text in the component selection panel at left and several other GUI parts. The font in the circuit canvas keeps unchanged.

dxhisboy, 2022-05-06:
And I also reproduced the problem on another screen, which is faked to be at 96 PPI.
[image]

[image]

xrandr output:
```
Screen 0: minimum 16 x 16, current 2560 x 1600, maximum 32767 x 32767
XWAYLAND0 connected primary 2560x1600+0+0 (normal left inverted right x axis y axis) 580mm x 360mm
```

fi11222, 2022-07-31:
@dxhisboy Could you tell me where to put the code you showed to fix the problem. I am using Ubuntu 20.04.4 LTS.

Thanks

3vi1, 2022-08-11:
I see this same thing.  The entire interface, menus and all appears to be at 2x zoom when the preferences are set to 1x, and there doesn't appear to be any way to scale down beyond that.  This is on Plasma (x11) desktop on Ubuntu 22.10 (Kinetic Kudu beta), using a 2560x1440 screen with no desktop scaling.

dxhisboy, 2022-10-07:
> @dxhisboy Could you tell me where to put the code you showed to fix the problem. I am using Ubuntu 20.04.4 LTS.
> 
> Thanks

I almost forget it, but I think the class name is correct, just use grep or other searching tools to locate the constructor of InstancePainter.

LAPSyLAB, 2022-10-16:
Hello,

have the same problem on 2550x1600 resolution. Have changed Windows system settings for Fonts size (from 175% to 150%),
restarted Logisim Evo and it helped... 
[image]

jimmy-zx, 2023-03-12:
Seems like the workaround will cause poking not to work, with the following report

> Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException: Cannot invoke "java.awt.Graphics.setFont(java.awt.Font)" because the return value of "com.cburch.logisim.comp.ComponentDrawContext.getGraphics()" is null

Adding a check before setFont would resolve this
```
if (this.context.getGraphics() != null) {
  this.context.getGraphics().setFont(new Font("monospaced", Font.PLAIN, 12));
}
```

lucianchauvin, 2024-02-27:
Hello. I am still having this issue and am not sure what the fix is?

maehne, 2024-05-16:
@lucianchauvin, @jimmy-zx, @LAPSyLAB, @dxhisboy, @3vi1, @fi11222: Could you please check with the [latest nightly build](https://github.com/logisim-evolution/logisim-evolution/actions/runs/9104282361) whether this issue persists? Please, use one of the native packages, which bundle JDK 21 instead of launching the JAR file directly with a potentially out-dated JRE installed via your operating system's package manager.

3vi1, 2024-05-19:
Looks much better now.  Tested with the AMD64 deb on an Ubuntu 24.10 (devel repos) base running Plasma 6.1 compiled from source.  Problem is fixed for me.

maehne, 2024-05-20:
@3vi1: Thanks for the feedback. I therefore close this issue as fixed.

@lucianchauvin, @jimmy-zx, @LAPSyLAB, @dxhisboy, @fi11222: Feel free to reopen in case the issue isn't resolved for you with the [latest nightly build](https://github.com/logisim-evolution/logisim-evolution/actions/runs/9104282361).

closed 2024-05-20 by maehne

maehne, 2026-08-21:
@Poly-Mentor: Please check with our [Nightly builds](https://github.com/logisim-evolution/logisim-evolution#nightly-builds). The issue should be fixed.