logisim-evolution/logisim-evolution #1453: Clock issues

TheDemonicStaff-t, 2022-05-04:
When I try using your clock on mac, it will not pulse.  The only way to have it run is by clicking it.

davidhutchens, 2022-05-04:
Could you please provide more information so we can try to reproduce this behavior.

What version of Logisim-evolution you are using?
Is it the .app or the .jar?
What version of Java do you have installed?
What version of Mac OS?

Note that some Java versions prior to Java 14 have issues with command keys that cause the use of cmd-k to fail. So if you are using an earlier version of Java, try upgrading to a more recent version.

AbiPrescott, 2023-11-24:
Hi. I'm having the same issue: clock won't pulse. Only works when I poke it. 

- I am on Ubuntu Server 22.04
- I am on Logisim Evolution v3.8.0.
- It was installed from snap store using command: sudo snap install logisim-evolution-snapcraft
- I checked my java version ($java --version), and this is what I get (which meets the requirement of having java 16 or higher):

     java 21.0.1 2023-10-17 LTS
     Java(TM) SE Runtime Environment (build 21.0.1+12-LTS-29)
     Java HotSpot(TM) 64-Bit Server VM (build 21.0.1+12-LTS-29, mixed mode, sharing)

- I'm guessing it is .jar? I found a logisim-evolution.jar in '/snap/logisim-evolution-snapcraft/46/logisim/logisim-evolution.jar'. 

Thank you :)

maehne, 2023-11-27:
@AbiPrescott: Are you sure that you selected `Simulate -> Auto-Tick Enabled` from the menu to activate the clock generation with the selected `Auto-Tick Frequency`?

AbiPrescott, 2023-12-04:
> @AbiPrescott: Are you sure that you selected `Simulate -> Auto-Tick Enabled` from the menu to activate the clock generation with the selected `Auto-Tick Frequency`?

@maehne yup

muhannad933, 2024-09-06:
same issue the clock does not work on my Mac mini m1

maehne, 2024-09-06:
@muhannad933: Please, provide a more precise description how to reproduce the issue! Which of the officially released most recent builds did you use? Please provide a test design exposing the issue in a ZIP archive and step for step instructions after loading it to reproduce the issue.

muhannad933, 2024-09-07:
I found a solution to the issue. The problem was resolved by enabling the correct simulation settings:

Go to the menu bar at the top of the screen.
Click on Simulate.
Ensure both Auto Propagate and Enable Ticks are checked.
After making these changes, the clock started working automatically without needing to toggle it manually.

Thank you for your assistance!

maehne, 2024-09-08:
Glad you were able to resolve your issue!

maehne, 2024-10-23:
@mercury64: In general, it is not good practice to ask questions in an already closed issue. You can choose the *Auto-Tick Frequency* in the *Simulate* menu. You can find more information in the online documentation, which is accessible via *Help* -> *User's Guide* -> *Logisim References* -> *Guide to Being a Logisim User* -> *Menu reference* -> *The Simulate menu*.