logisim-evolution/logisim-evolution #1454: Request for adding iCore fpga by Gingko.

gnudles, 2022-05-13:
I got the iCore 1 from a friend, I wonder if anyone else has it, and can help

gnudles, 2022-05-14:
Thanks. I got the board from a friend, so I don't have the documentation of the borad. I'll try to get it.

gnudles, 2022-05-29:
[icore.rev0.pdf](https://github.com/logisim-evolution/logisim-evolution/files/8794208/icore.rev0.pdf)
This is the pin scheme for that board. I uploaded it cause it might be useful for someone in the future

maehne, 2022-06-01:
@gnudles: Indeed, these schematics plus data sheets for the used components should allow you to create a suitable board configuration file with the board editor. You can focus on the components connected to the EP4CE6E144 symbols. As Logisim-evolution already supports some boards with Cyclone IV FPGAs, you can use them as reference to see how certain information should be formatted. Besides the 25 MHz clock, you could start with mapping the buttons, LEDs, and GPIOs. Once you have draft configuration, you could open a PR against our develop branch and ask for feedback.