logisim-evolution/logisim-evolution #1455: [Feature Request] Change keybindings

TornaxO7, 2022-05-13:
Hello! I couldn't find a way to change the default keybindings for example to change to "Edit selection and wires"-mode.
It would be nice to map them to something else than using `Ctrl`+`<num>` :)

MarcinOrlowski, 2022-05-13:
Keybindings are currently not configurable.

maehne, 2026-08-21:
I agree, but would not limit it to the first 10 toolbar icons, but also would offer to configure shortcuts for those, which currently don't have one assigned (and by default should be left unassigned).

IMHO, the usability of the toolbar could be improved. Only the first 6 entries, I use regularly. Between the hand tool and selection tool, I typically switch using the keyboard shortcut. It might make sense to make the toolbar configurable as in many modern GUI applications, but this is outside the scope of this issue.