logisim-evolution/logisim-evolution #1457: ROM components are affected by the Randomize option

prlaba, 2022-05-18:
The contents of a ROM component should not be affected by the _Memory elements start in unknown/random state_ project option (off by default).

Enabling that option appears to affect ROM components in two ways:

1) If you use the _Edit contents_ option to edit a ROM's contents while the Randomize option is enabled, clicking on a cell will immediately randomize that cell and some of the nearby cells (bizarre).  Exiting the option will randomize some or all of the cells (I couldn't figure out the exact behavior).

2) If you close a circuit containing a ROM, with the Randomize option enabled, when you next open the circuit all 00 cells _beyond the last non-00 cell_ in the ROM are randomized.  If the ROM is blank (all 00's), all cells except the first will be randomized.  If the last cell in the ROM is not 00, _all cells will be preserved_.

Case 2 is particularly nasty because it can cause some parts of some ROM components to be randomized but not others, depending on their contents.
 
Although ROM components are considered a "memory element", their contents should never be affected by the Randomize option; only by the _Edit_, _Clear_ or _Load_ options.

I'm forced to keep the _Memory elements start in unknown/random state_ option disabled because of this bug.

Case 2 of this bug appears to be the result of how L-E stores a ROM's cells in the XML .circ files.  Only cells up to and including the last non-00 cell are stored; L-E fills in the remaining cells with 00's (or random values if the Randomize option is set).

There's no reason why _all_ of a ROM's cells can't be specified in the XML file, since it already uses <count>*<value> notation to indicate a contiguous group of cells with the same values (e.g., 35*ff indicates 35 contiguous cells with value ff).  A blank 64K x 8 ROM can be fully specified with a single 65536*0 value. L-E currently specifies a blank ROM with a single 0, which explains why the Randomize option causes all remaining cells to be randomized.

Specifying all of a ROM's cells in the XML file would at least fix the case 2 bug.  The right fix is for the Randomize option to not apply to ROM components.

This bug doesn't exist in the original Logisim as it has no Randomize option.

maehne, 2022-05-18:
Thanks for the detailed bug report! I agree with your proposals how the ROM component should behave. Currently, our time for Logisim-evolution is limited due to the nearing end of the term and the upcoming exam period. PRs are welcome!

prlaba, 2022-05-18:
Sorry, the GitHub editor seems to have eaten some of my asterisks.  Here's the confusing paragraph again with 'x' in place of asterisk.

"There's no reason why all of a ROM's cells can't be specified in the XML file, since it already uses [count]x[value] notation to indicate a contiguous group of cells with the same values (e.g., 33xff indicates 35 contiguous cells with value ff)."

BFH-ktt1, 2022-08-07:
Fixed by my latest PR. Please reopen if problem persists

commit 5d7fce9d (2022-08-07): This PR fixes the Random filling of a rom if the startup-option fill
memory with random is selected. this fixes #1457
commit ac3e8fe1 (2022-08-07): Merge pull request #1507 from BFH-ktt1/rombug

This PR fixes the Random filling of a rom if the startup-option fill memory with random is selected. This fixes #1457
closed 2022-08-07 by BFH-ktt1