logisim-evolution/logisim-evolution #1467: Logisim-HolyCross appearance settings not saved

prlaba, 2022-06-17:
If you use a component from a library and assign it the Logisim-HolyCross appearance setting, the setting is not saved.   When you close and reopen the master circuit, the component's appearance reverts back to Logisim-Evolution, messing up all the connections to/from the component.

maehne, 2022-06-17:
Your issue might be related to the still open issue #1379. Unfortunately, our time for Logisim-evolution is currently limited, as we first need to wrap up the spring term.

prlaba, 2022-07-04:
I discovered that the appearance you assign to a circuit is the appearance that "sticks" when you use it in a master circuit.

It would be better if you could change the appearance of a subcircuit in the master circuit, consistent with how a component works.

maehne, 2022-10-08:
@prlaba: PR #1380 fixed the potentially related issue #1379. Could you please check whether this issue still is present in Logisim-evolution 3.8.0?

chuckb, 2023-01-29:
Yes, I will try. Sorry for the delay.

maehne, 2023-01-30:
@chuckb: Thanks for these detailed steps to reproduce the issue!

@BFH-ktt1: I can confirm the bug!

chuckb, 2023-01-30:
I may have some time in the next few days to take a stab at fixing it...

maehne, 2023-01-31:
>I may have some time in the next few days to take a stab at fixing it...

@chuckb: That would be great!

maehne, 2025-06-20:
This seems not to be reproducible any more on the `main` branch, as you cannot change any more the appearance of the placed component.

@chuckb: Can you confirm this? If yes, can this issue be closed as resolved?

maehne, 2025-09-05:
I am closing this issue as resolved, as I cannot reproduce it with HEAD and lacking activity. Feel free to reopen if the issue persists in our Nightly builds.

closed 2025-09-05 by maehne