logisim-evolution/logisim-evolution #1471: Chrome file downloaded has .circ extension even though it of different extension type

addtogether, 2022-06-24:
I was downloading a zip file but instead of it downloading as zip it downloaded as a .circ file. Don't know what's causing this issue but i am using logism-evolution version 3.7.2

MarcinOrlowski, 2022-06-25:
We had similar report already in the past (#1449) but as stated in the linked thread we will  likely not be actively investigating due to limited bandwidth. If you will have any reproducible case and/or evidence that made you be sure to point your finger at LSe, please share it with us. 

* #1449

R3dst0ne, 2022-06-25:
The same problem got discussed in [1449](https://github.com/logisim-evolution/logisim-evolution/discussions/1449).
Since multiple people are effected, it might be worth looking into it.

Do you mind sharing information about your system?
How did you install LSe?
What OS, browser, Java (all with version numbers) do you use?

addtogether, 2022-06-25:
> @MarcinOrlowski was faster than me. ^^'
> 
> As stated my @MarcinOrlowski we will likely not be actively investigating. But it might be worth sharing your system information, in the case someone wants to take a look into this. How did you install LSe? What OS, browser, Java (all with version numbers) do you use?

For installation i used .msi file from https://github.com/logisim-evolution/logisim-evolution/releases

As for os i am using Windows 10 (19044.1766)
Browser as mentioned in heading i am using chrome (103.0.5060.53)
For java jdk 14.0.2

addtogether, 2022-06-25:
> We had similar report already in the past (#1449) but as stated in the linked thread we will likely not be actively investigating due to limited bandwidth. If you will have any reproducible case and/or evidence that made you be sure to point your finger at LSe, please share it with us.
> 
> * [Logisim turns downloaded files into .circ files? #1449](https://github.com/logisim-evolution/logisim-evolution/discussions/1449)

As for the evidence when i had logism-evolution installed on my system and tried to download the file from file server it downloaded as .circ not as .zip which was the original extension of the file.
Then i went ahead and uninstalled logism-evolution then tried to download same file again and this time it downloaded as .zip
and for your reference i reinstalled logism-evolution and did the same thing again and it downloaded as .circ file.

This is how i am saying that it was logism-evolution which changed the extension.

maehne, 2022-06-25:
@addtogether: Thanks for the detailed replies. Besides the Chrome version, do you use any Chrome plug-ins or other tools, which might interfer with the download process? Do you observe the same behaviour, if you use just the [Logisim-evolution JAR archive](https://github.com/logisim-evolution/logisim-evolution/releases/download/v3.7.2/logisim-evolution-3.7.2-all.jar) with a locally installed JDK? Does the problem happen if you build yourself the MSI from the sources?

We never got reports in this direction from our students neither were able to observe it ourselves. Can you share a link / several links, where you observed this behaviour? On which kind of web sites does it happen? Without a way to reproduce it, we cannot look into the cause or provide a fix.

MarcinOrlowski, 2022-06-26:
I think that what is probably worth investigating is Windows registry extension/file type mapping. `*.circ` is just XML text file.

maehne, 2022-06-26:
@MarcinOrlowski: I agree that this might contribute to the issue: Some browsers tend to render `.circ` files as XML files in the web browsers instead of downloading them.

MarcinOrlowski, 2022-06-26:
Another thing for anyone experiencing this problem - can you please try installing **older** versions of LSe to see if they trigger this too? And if not then which one stared doing so -> see app's About dialog and look for "Build info" (including git hash) at the end of the scroll text and share it here?

[image]

Please hare both lines, compile time and "branch/hash" strings - esp. hash is crucial here.

MarcinOrlowski, 2022-06-26:
> For java jdk 14.0.2

@addtogether how come? LSe requires JDK 16 or higher.

MarcinOrlowski, 2022-06-26:
Ah, never mind. I just realized that bundles we create are self contained.

addtogether, 2022-06-27:
> @addtogether: Thanks for the detailed replies. Besides the Chrome version, do you use any Chrome plug-ins or other tools, which might interfer with the download process? Do you observe the same [...]

For chrome plugins i am not using any plugin that affects download.
I am using the .msi installer for installing as for other i haven't tried it (or should i say i don't know how to install using jar :( )
As for websites on which this happened was an anime download website which provides .zip files for various series using their file server.

addtogether, 2022-06-27:
A correction to my statements above
As stated it happened on an anime website but what i found now was this issue occurs on .rar files.
I downloaded .zip files from the same file servers but that didn't got turned into .circ but .rar files did

MarcinOrlowski, 2022-06-27:
Any chance you could install browser extension like Live Headers and then paste the response headers that came from server with the file? I mostly mean mime type but please paste them all.

https://chrome.google.com/webstore/detail/live-http-headers/ianhploojoffmpcpilhgpacbeaifanid

addtogether, 2022-06-30:
> Any chance you could install browser extension like [Live HTTP Headers](https://chrome.google.com/webstore/detail/live-http-headers/ianhploojoffmpcpilhgpacbeaifanid) and then paste the response headers that came from server with the file? I mostly mean mime type but please paste them all.

Here you go:

GET /Vongola/Ascendance%20of%20a%20Bookworm%20Part%202/[AnimeKaizoku]%20Ascendance%20of%20a%20Bookworm%20Part%202%20720p%20Dual%20Audio%20(LostYears)%20[dedsec].rar?r=rNKDAnkJ&tid=MEFIV1pnWU1ubno3MlVrOVBWQQ==&t=5b145&e=1656074 HTTP/1.1
Host: kuro-vongola.animekaizoku.com
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/avif,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3;q=0.9
Upgrade-Insecure-Requests: 1
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/103.0.0.0 Safari/537.36

HTTP/1.1 200 OK
Accept-Ranges: bytes
Content-Length: 1684421611
Content-Type: application/octet-stream
Date: Thu, 30 Jun 2022 13:41:30 GMT
Last-Modified: Sun, 27 Sep 2020 12:55:22 GMT
Server: nginx
Vary: Accept-Encoding
Via: 1.1 vegur

lautis0503, 2022-10-08:
Same problem here.
This is Windows 7 SP1, with `logisim-evolution-3.5.0.msi` installed last year.
Download files with extension .zip or with no extension in Chrome, would be saved to .circ files.
Unintall logisim-evolution solved the problem.

maehne, 2022-10-08:
@lautis0503: Can you provide links to web sites/downloads, which cause this issue? We have not yet been able to reproduce the issue on our side, which makes it difficult for us to find an appropriate fix.

AlexisGross, 2022-10-09:
Original message:
Tengo el mismo problema, todos los archivos me los descarga con la extensión .circ, desinstalando Logisim se soluciona, uso Microsoft Edge como navegador y no tengo ninguna extensión instalada en el navegador.

Message translated by Google Translate:
I have the same problem, all the files are downloaded to me with the .circ extension, uninstalling Logisim is solved, I use Microsoft Edge as a browser and I do not have any extension installed in the browser.

lautis0503, 2022-10-09:
@maehne Not a single site, but IIRC all the downloads with .zip extension would be saved to .circ. (There was once that another extension had the same problem, but I did't remember what it is.)
And when I downloaded files from Google Drive, in which I saved some text files with no extension, they were also saved to .circ files.

I find another clean computer, which had not installed logisim or logisim-evolution before. The OS is also Windows 7 SP1. I tried to download .zip files both before and after the installation of [logisim-evolution-3.8.0-x86.msi](https://github.com/logisim-evolution/logisim-evolution/releases/download/v3.8.0/logisim-evolution-3.8.0-x86.msi), it all works fine.

Then I'm aware of one difference between the two computers (my computer, and the clean computer).

- In my computer, the .zip file are handled with Windows Explorer. If I open Control Panel -> Default Programs -> Associate a file type or protocol with a program, the .zip file type are not there.
- In the clean computer, the .zip file are associated to WinRAR, and the same in Control Panel.

Hope this would help. And you can tell me what to do to help solving this problem. Thanks.

dbh201, 2022-11-09:
I believe this may have to do with the mime-type setting in jpackage. A lot of servers report files with no extension (or erroneously for files with certain filetypes) as `application/octet-stream`. Now, this commit seems to link the mime-type with the file extension: [see the jpackage files in this commit.](https://github.com/logisim-evolution/logisim-evolution/commit/a400b72fd678541363339ab298898288fe092cf4)

I don't know why the Chromium engine (which is used by Microsoft Edge too, among others) would append extensions to files which already have extensions - I couldn't replicate that issue, even when setting `Content-Type` to `application/octet-stream` for a test download. However, I've confirmed that the engine will append extensions to extension-less files based on the registry key in HKEY_CLASSES_ROOT/MIME/Database/Content Type/`application/octet-stream` which is created/edited during install from the .msi.

Maybe the mime-type should be removed, or changed to something like `application/logisim-evolution-circ-xml` to avoid conflicts.

Dudejoe870, 2022-11-12:
This also happened to me while downloading NVidia tools from their developer website.

maehne, 2022-11-15:
@dbh201: Thanks for the detailed analysis of the problem! `.circ` files are XML documents. So, we could consider using simply `application/xml`. As far as I know, new MIME types need to be centrally registered and your proposal doesn't seem to be compatible with the MIME standard, as Logisim-evolution's MIME type should probably be part of the [vendor tree](https://en.wikipedia.org/wiki/Media_type).

@Dudejoe870: Could you please provide the exact link to the developer website and name of the package, so that we can try to reproduce the issue on our side and test any fix based on @dbh201 findings?

maehne, 2022-11-15:
@lautis0503: Thanks for sharing your observations!

dbh201, 2022-11-15:
My previous suggestion was a bit absent minded: `application/x-logisim-evolution-circ+xml` would be my suggestion for MIME type.

My concern about `application/xml` is that it's going to tack `.circ` extensions on XML files for some people. Technically, a `.circ` file is an [XML-based media type](https://www.rfc-editor.org/rfc/rfc7303.html#section-9.6).

An application needs to be able to open all files of that MIME type to be adherent to the RFCs. LSe cannot open all `application/xml` files, but XML parsers *can* open all files that are XML-based. 

So, three options that I can see:
 - Leave the MIME type line absent in the jpackage files
 - Use an experimental MIME type (popular choice for small vendors with esoteric filetypes)
 - [register a MIME type](https://datatracker.ietf.org/doc/html/rfc6838#section-5)

Dudejoe870, 2022-11-19:
@maehne: It happened to pretty much every single installer on their website but I was specifically trying to install Nsight at the time. Here's where it happened to me: https://developer.nvidia.com/gameworksdownload#?dn=nvidia-nsight-graphics-2022-5 (You may have to make an NVidia developer account to access it I'm not sure). It's an MSI file.

Dudejoe870, 2022-11-19:
Oh also I should note, I am using Opera GX. So I guess it might be happening to any Chromium based browser considering the original post is about Chrome.

Khitiara, 2022-12-01:
vivaldi for me, all mp4 files. definitely seems to be on chromium based stuff

djmips, 2023-01-28:
mhm - I have installed logisim v3.7.2 Build info: Compiled 2021-11-09T10:27:32+0100 master/26482a19 Java HotSpot(TM) 64-Bit Server VM v16.0.2 Oracle Corporation. And for me it's Chrome and saving PDF and MP3 files a .circ.  I'll try the fix mentioned in previous comment.

man-of-eel, 2023-03-31:
Logisim has something strange going on with its MIME type registries on Linux as well. A lot of random files seem to get detected as "Logisim circuit file" instead. Been trying to fix it (personally) to no avail. I haven't had this specific issue with random files turning into `.circ`s though

maehne, 2023-04-04:
@man-of-eel: Thanks for the feedback. Have you tried @dbh201's suggestion to remove the `mime-type` line from the `file.jpackage` files in `jpackage/...` and then building/installing the packages?

Klemen2, 2023-06-06:
Yes, the above suggestion works. I assume that the bug is caused because many file servers do not send filename in response header and have the name only in the url. This can hopefully be reproduced with the bellow code while clicking on the "Download PNG" button. The xml mimetype on the otherhand does not cause this issue, or at least I cannot make it to rename xml extension.
```python
from flask import Flask, Response, render_template_string
from PIL import Image
import io

app = Flask(__name__)

xml_content = '''<?xml version="1.0" encoding="UTF-8"?>
<root>
  <message>Hello!</message>
</root>'''

image_data = b'\x89\x50\x4E\x47\x0D\x0A\x1A\x0A\x00\x00\x00\x0D\x49\x48\x44\x52\x00\x00\x00\x01\x00\x00\x00\x01\x08\x02\x00\x00\x00\x90\x77\x53\xDE\x00\x00\x00\x0C\x49\x44\x41\x54\x08\xD7\x63\xF8\xCF\xC0\x00\x00\x03\x01\x01\x00\x18\xDD\x8D\xB0\x00\x00\x00\x00\x49\x45\x4E\x44\xAE\x42\x60\x82'
image = Image.open(io.BytesIO(image_data))

html_template = '''
<!DOCTYPE html>
<html>
<head>
  <title>Web Server</title>
</head>
<body>
  <h1>Web Server</h1>
  <button onclick="showXML()">Show XML</button>
  <button onclick="downloadPNG()">Download PNG</button>
  <script>
    function showXML(){ window.open('/hello.xml', '_blank'); }
    function downloadPNG(){ window.open('/red_pixel.png', '_blank'); }
  </script>
</body>
</html>
'''

@app.route('/')
def index():
    return render_template_string(html_template)

@app.route('/hello.xml')
def show_xml():
    return Response(xml_content, mimetype='application/xml')

@app.route('/red_pixel.png')
def show_png():
    image_bytes = io.BytesIO()
    image.save(image_bytes, format='PNG')
    image_bytes.seek(0)
    return Response(image_bytes, 
                    mimetype='application/octet-stream'
                    # if the server sends with this in the header it will download normaly
                    # headers={'Content-Disposition': 'attachment; filename="red_pixel.png"'} 
                    )

if __name__ == '__main__':
    app.run()
```

maehne, 2023-07-07:
I am closing this issue as fixed since PR #1765 got merged. Feel free to reopen it if the problem should persist!

closed 2023-07-07 by maehne

Klebestreifen, 2024-06-06:
Can you please publish the patched version in the releases? I installed the buggy version of 2022 as it is listed as the latest and now I have the problem too.

maehne, 2024-06-18:
@Klebestreifen: We agree that a release is overdue. Unfortunately, @BFH-ktt1 has been quite busy with other responsibilities. IMHO, we should strive for a new release before the autumn term starts mid September.

@BFH-ktt1: What do you think?

In the meantime, the nightly builds are your best option.