logisim-evolution/logisim-evolution #1477: Allow setting Input Pin component's initial state

prlaba, 2022-07-15:
When a circuit is opened or the simulation is reset, all of the circuit's input pins are reset to low (0).  This can cause errors if the pin is intended as an active-low input, as it causes the pin to be initially active.

This is not usually a problem, as most components use active-high inputs, even though their real-life counterparts often use active-low inputs (e.g., the Controlled Buffer component uses an active-high Enable input; all bus buffer IC's I've ever seen use an active-low Enable pin).

I often create custom sub-circuits to replace L-E's native components, so I can get the polarity of the component's inputs correct.  But that causes problems because any active-low inputs are initialized to 0 (active).

It would be great if L-E had an _Initial State_ attribute for its Input Pin component, that would allow the pin's state to be set either high (1) or low (0) initially.  It's default would be low (0), making the feature backward compatible.

Thanks.

BFH-ktt1, 2022-08-06:
I find this a very good idea and see how to implement it.

BFH-ktt1, 2022-10-03:
Is in V3.8.0. Please reopen if the problem persists.

closed 2022-10-03 by BFH-ktt1