logisim-evolution/logisim-evolution #1478: No SEL pins for ROM and RAM

prlaba, 2022-07-16:
The original Logisim ROM and RAM components had SEL pins that had to be set (1) to select/enable the device.

In fact the Logisim Evolution Help reference page still describes the original Logisim RAM and ROM components, including their SEL pins (that's another issue).

The RAM component does have a C1 "clock" input pin that acts somewhat like a SEL pin, but it only enables/disables the RAM's store/write function, not its load/read function.  The ROM component has only address pins and data (output) pins.

The SEL pin did exactly what you would expect.  For the RAM component, when SEL is 0 (device disabled), both the store (M1 pin) and load (M2 pin) functions are disabled (those pins have no effect) and the data outputs are high-Z.  For the ROM component, when SEL is 0 its data outputs are high-Z.

It's not clear to me why L-E removed those SEL pins, given that every RAM or ROM IC I've ever seen has at least one select/enable pin.

Without those pins I'm forced to add extra logic around the RAM and ROM components to perform the missing SEL function.  And if I need to use multiple RAM and/or ROM components, I either have to add that logic around every instance, or create a subcircuit, which hides most of the useful functionality of the native RAM and ROM components (address grid, currently selected address, right-click to edit, load, etc.).

Please consider (re)adding the SEL pin to the ROM and RAM components.

Thanks.

BFH-ktt1, 2022-08-06:
The sel pin was removed in favor of the WE and OE pin. Although many "external" memories have a nCS pin, FPGA memories do not have them and providing this functionality only takes an and-gate and an inverter. Hence the "overhead" is limited.

prlaba, 2022-08-06:
Understood, the OE pin on the RAM component effectively provides the SEL function.

But the ROM component has no OE pin.  You have to work around the missing OE pin by passing the ROM's outputs through a Controlled Buffer component, controlled by an OE signal.

Every EEPROM and ROM component I have ever worked with has a CE or OE pin.  When inactive, the ROM's outputs are high-Z.

This matters if your design includes multiple ROMs with shared output/data pins - tying some or all of the pins together and using the ROMs' OE pins to enable one of the ROMs and disable the others.

Please consider adding an OE input pin to the ROM component.