logisim-evolution/logisim-evolution #1479: Logisim Evolution documentation out of date

prlaba, 2022-07-16:
This is a known issue, but I wanted to highlight the fact that the documentation for L-E's RAM and ROM components still describe the original Logisim components, not the L-E ones.

This is especially painful for L-E's RAM component, which underwent significant changes from its original Logisim component.  It now has many more attributes, some of which are cryptic and non-intuitive (it took me a while to figure out what the various attributes did).

If you decide to upgrade L-E's documentation, please put the RAM component at the top of the list.

codeBodger, 2022-08-12:
Could you please briefly write up what you figured out?  It would help anyone who's struggling and make rewriting the docs easier.

prlaba, 2022-08-12:
Sure, I'll send you some text.

Paul

mbaillif, 2023-07-20:
Hello

I need help to proofread the new documentation of the Memory library pages (10).

They are available on this link   https://www.baillifard.com/logisim/ 
and also on   https://github.com/mbaillif/Logisim-Evolution-documentation

I've made a lot of changes to the layout and some text. It's really a lot of work.

The Library libs status link gives direct access to the current status of the documentation.
If you click in one of the boxes, the pages can be accessed directly.

If you're really motivated, you can start proofreading all the pages with a _V.

This is a first attempt. There's still a lot to do.

If you're not too comfortable with html or Github, you can send me a Word or OpenOffice document (copy and paste from the web browser) with the corrections clearly visible.

maehne, 2023-08-28:
> I need help to proofread the new documentation of the Memory library pages (10).

Thanks, @mbaillif, for all your work on the documentation! I will try to allocate some time for proofreading, but cannot make any promises, when I'll be able to work on it. The autumn term is approaching quickly…

maehne, 2026-08-21:
I agree and close this issue as resolved.

closed 2026-08-21 by maehne