logisim-evolution/logisim-evolution #1483: Improve German localisation strings

maehne, 2022-07-17:
The following screen shots serve as reminder to improve some erroneous German localisation strings:

- [ ] [image]

[image]

[image]

[image]

maehne, 2022-08-23:
- [ ] "Anti-Aliasing" -> "Antialiasing" (Voreinstellungen -> Layouteditor)

MarcinOrlowski, 2022-11-04:
@maehne this is tagged `3.8.0`. Can you please update the target?

closed 2025-09-06 by BFH-ktt1