logisim-evolution/logisim-evolution #1493: Evolution runs slower than Logisim even at max speed on my pc, is that normal?

agoodfellow123, 2022-08-01:
Hello there, I have been using logisim and logisim evolution for a few days on my PC. I'm sorry if it's supposed to be like this but at first glance I thought evolution supports higher speeds than logisim but it's quite the opposite on my PC. I installed logisim evolution from flatpak and I'm on manjaro Linux. I don't know what info I should share so please ask me if you need additional info.

davidhutchens, 2022-08-02:
Thank you for the report.

We had a deep look into a slowdown introduced between 2.14.4 and 2.14.6 in issue #216. Our conclusion was that the thread synchronization in early versions had problems that would rarely but occasionally cause crashes. Those issues were fixed at the cost of slowing down the simulation.

So it is normal for the simulation to run slower in logisim evolution, but it also runs more reliably.

agoodfellow123, 2022-08-02:
I see, thank you for the explanation. Is there a plan to fix this or at least remove the misleading upper speeds from the menu?

davidhutchens, 2022-08-02:
We have some thoughts about speeding up the simulation, for example, #786. But they require careful evaluation of the simulation engine and are therefore difficult to implement. Our available time to work on it is limited, and other pressing features and bug fixes take precedence. Work (analysis or PRs) in this direction is welcomed.

As for the upper speed limits, the actual upper limit is reached when your particular hardware is maxed out. That differs from one system to another and from one circuit model to another. For example, I have some complicated circuits that can only get 50 Hz or so on my hardware. So the menu actually means "run no faster than this speed" rather than "run at this speed".

agoodfellow123, 2022-08-02:
Ah I see, that makes sense. Thank you.

BFH-ktt1, 2022-08-05:
Can this issue be closed?

agoodfellow123, 2022-08-05:
I see no problems. I'd close it myself but I don't know if that's ok.

maehne, 2022-08-07:
I close this issue as a duplicate of issue #786.

closed 2022-08-07 by maehne