logisim-evolution/logisim-evolution #1494: If you use a dipswitch and set it to have 16 switches, number 13 keeps giving an error and all the wires connected to it shows red

Deadlyfeat, 2022-08-01:
I have already tried redoing the project and it still shows up anyways

davidhutchens, 2022-08-02:
I cannot reproduce this in our current develop branch. Could you please tell us what version of LE you are using?

Also, could you please attach a .circ file that shows the problem? You will need to zip it first, then drag it into here. If it is complex, it would help if you could remove parts of the circuit to isolate the problem and post a simple .circ to show it. Please post the whole thing if you can't easily simplify it.

maehne, 2022-08-07:
@Deadlyfeat: Please, answer the questions by @davidhutchens and provide a reproducer. Otherwise, we cannot help and will close this issue.

Deadlyfeat, 2022-08-07:
I was designing an 8Bit-Adder by the channel crash course
Computer science and the Adder had 16 inputs and when I would activate the 13th switch it would make the wire that's connected to it red, and nothing i did changed the problem. But when I made it into a single module it solved the problem.

Deadlyfeat, 2022-08-07:
And it is the most recent version

Deadlyfeat, 2022-08-07:
it now works for some reason

closed 2022-08-07 by Deadlyfeat

Grant-fyx, 2025-07-24:
I think it randomly selects a pin to output ERROR.

[image]

davidhutchens, 2025-07-24:
@Grant-fyx: My guess is that this issue is caused by the simulator leaving hot spots on the canvas when rotating a component. This is fixed in the current main branch. You might try the Nightly build. But that isn't really necessary, because if I am right, saving and opening the circuit will fix it. Or you can fix it by doing a Simulator Reset from the menu.