logisim-evolution/logisim-evolution #1496: Logisim 3.6.1 not detecting 7seg as output

AvaDanialy, 2022-08-04:
I tried to synthesize and download my design onto the DE1_SoC board but at the execute step it doesn't detect the 7seg and I cannot assign the output correctly to a 7seg, and my only options are LEDs. Can anyone guide me what to do?

BFH-ktt1, 2022-08-05:
The problem here is that the 7-segment displays on the DE1_SOC do not have a decimal point. There are two ways to solve this problem:
1) Disable by the properties of the 7-segment display the decimal point, and they are now "mapable".
2) Change to logisim 3.7.2 (see the release page). This version includes an improved mapping scheme where you can map for example even a LED on a single segment of a 7-segment display.
Hope this helps.

AvaDanialy, 2022-08-05:
Thanks! I currently cannot use any newer version, but I'll try the first option! How can I get access to the xml file (pin assignment) for this board to add to Logisim? Just want to make sure that my version isn't problematic.

MarcinOrlowski, 2022-08-05:
> I currently cannot use any newer version

May I ask what stops you from upgrading?

AvaDanialy, 2022-08-06:
I'm currently trying to do this on a computer that cannot take any upgrades or updates.

maehne, 2022-08-06:
@AvaDanialy: Please report if the first option is working for you.

If you are not able to update your installed Logisim-evolution version, we can only provide very limited support.

BFH-ktt1, 2022-08-19:
Can we close this issue?

AvaDanialy, 2022-08-19:
> You don't often get email from ***@***.*** Learn why this is important<https://aka.ms/LearnAboutSenderIdentification>

No it didn’t work unfortunately.

maehne, 2022-08-22:
@AvaDanialy: Thanks for the feedback! Could you please try with the most recent release 3.7.2 or even better the HEAD of the `develop` branch? If it's not working there, we can work on the issue. Otherwise, we'll have to close this issue, as we don't have the resources to maintain older releases as well.

AvaDanialy, 2022-08-22:
I understand Torsten. Thank you so much for helping me with this issue. Unfortunately I don’t have the resources necessary to try it with the recent release…

maehne, 2022-08-22:
@AvaDanialy: Thanks for the feedback! I'll close the issue for now, as we cannot reproduce the issue on our side without having access to the board. Please, feel free to reopen this issue once you can confirm that the issue persists in the most recent release/develop branch.

closed 2022-08-22 by maehne