logisim-evolution/logisim-evolution #1503: Remove unmaintained translations

MarcinOrlowski, 2022-08-05:
We should remove all unmaintained translations from LSe as these apparently cause more harm than help.

---

I launched this software and set the language to Chinese, but the toolbox showed a lot of "boxes", which is referred to the lack of font. Maybe my system does not have thois font. Can you help me?

_Originally posted by @fumarate in https://github.com/logisim-evolution/logisim-evolution/discussions/1490_

MarcinOrlowski, 2022-08-05:
I am going to see what percentage of translations are missing from each of non English strings. We should set a threshold (say 75%?) and hide/remove translations with lower coverage. Alternatively, we may consider using any of the community driven translation services and upload our files there for anyone to contribute. But still, anything below the threshold should be gone as we cannot assure any quality.

MarcinOrlowski, 2022-08-05:
I am going to see what percentage of translations are missing from each of non English strings. We should set a threshold (say 75%?) and hide/remove translations with lower coverage. Alternatively, we may consider using any of the community driven translation services and upload our files there for anyone to contribute. But still, anything below the threshold should be gone as we cannot assure any quality.

@maehne @BFH-ktt1 @davidhutchens @zdimension

R3dst0ne, 2022-08-06:
The `percentage of translation` is an interesting metric, but I don't think it should be the only one.
E.g. the SOC and FPGA parts have together roughly the same amount of translations as the standard libraries, but most people don't care about FPGA and SOC. For most people, the translation state of the standard libraries and GUI is important. If a language has an excellent coverage in those two, I don't think it should be removed only because it lacks in less often used areas.

---
> We should remove all unmaintained translations from LSe as these apparently cause more harm than help.

The Chinese translation is quite new. It was introduced in #1282.
@fumarate's problem has to do with the font, not the state of the translation.

maehne, 2022-08-07:
I think removing unmaintained localisations is not the solution. If someone wants to improve a localisation, he cares about, he will appreciate any visible starting point, as bad its state may be.

That Logisim-evolution becomes unusable in some localisations is primarily caused by missing font glyphs for languages with non-Latin alphabets. IMHO, we should rather investigate, whether we can switch the fonts used in Logisim-evolution's GUI to some family with better Unicode glyph coverage, e.g., Google's Noto font family. However, using the system's default GUI fonts ensures typically a better integration into the desktop environment and if I am not mistaken, all major platforms (Windows, macOS, Linux with GNOME, KDE Plasma, etc.) improved considerably in this area over the past years.

R3dst0ne, 2023-02-05:
Can this issue be closed?

closed 2023-02-10 by maehne