logisim-evolution/logisim-evolution #1510: Provide invert feature for component inputs

prlaba, 2022-08-06:
Most L-E components use active-high input pins for enabling, selecting, etc.

Oftentimes that polarity is opposite the polarity of the pin found on the corresponding physical components.

For example, the WE and OE pins on L-E's RAM component are active-high, while the corresponding pins on all static RAM ICs are active-low.

While it's easy enough to add inverters to reverse the polarity of signals tied to those pins, it would be better to be able to indicate these pins' polarity as as attribute of the component itself.  Marking a pin active-los would add an active-low "bubble" to the pin, the same way the inverted inputs appear for the AND, OR and other gate components.

This should apply to these components (there may be others):

RAM (WE and OE pins)
ROM (OE pin, if it existed)
D, JK and SR Flip-Flops (Set and Reset pins)
Multiplexer (Enable pin)
Demultiplexer (Enable pin)
Decoder (Enable pin)

(For the Demultiplexer and Decoder components, it would be great to be able to mark their outputs as inverted as well; most Demultiplexer and Decoder ICs have active-low outputs)

Doing this would eliminate the need for inverters to reverse the active-high inputs' polarity.  More importantly, it would allow L-E circuits to more closely reflect the physical components used in those circuits.

maehne, 2022-08-06:
Pull requests in this direction are welcome!

prlaba, 2022-08-06:
I wish I were skilled enough w Java to do this.

maehne, 2022-08-07:
@BFH-ktt1: Would you maybe like to take this up, as you addressed the related issue #1477 with PR #1506?

figurantpp, 2023-08-26:
With regards to multiplexes and the like, if we did include a circle indicating that the select is negated, given a multiplexer like the one attached in the picture, would it be ok to move the select input downards by one unit to accommodate the circle?
[image]

davidhutchens, 2023-08-26:
I am not sure it makes sense to negate the select input to the multiplexer since the select input can be many bits wide and is treated as an integer value. Negating all of them seems strange.

The request was for a negation of the **enable** pin, not the **select** pin.

figurantpp, 2023-08-27:
This makes sense, thanks for the clarification.

maehne, 2023-08-28:
> I am not sure it makes sense to negate the select input to the multiplexer since the select input can be many bits wide and is treated as an integer value. Negating all of them seems strange.
> 
> The request was for a negation of the **enable** pin, not the **select** pin.

I agree with @davidhutchens. Nevertheless, just for the record: It is acceptable to offset the pin location when adding the circle, as it is already the case for the basic logic gates, which allow negating their inputs.

prlaba, 2023-08-31:
In general it would be very useful to be able to invert any enable, set or reset input of a component, to avoid having to use inverters.  This would apply to memories, flip-flops, counters, etc.  Another possible candidate would be the CLK input of any component.

Inverting an input would add a bubble to the displayed component’s input, similar to how inverted inputs appear on the logic gate components.

Paul

maehne, 2026-08-20:
Thanks @hewzhew for volunteering to pick up PR #1855 and bring it to a state allowing it to be merged! The approach you propose sounds good to me. If the activity for all enables are controlled via a single attribute, I personally think that the PR could also considerate byte/line-enable polarity like @davidhutchens suggested.

@BFH-ktt1: Do you agree? Any additional thoughts?