logisim-evolution/logisim-evolution #1514: trying to overwrite a rom when saving one as a file has a placeholder message

j0912345, 2022-08-15:
this is what I mean:
[image]
[image]
I'm running windows 10

BFH-ktt1, 2022-08-23:
I can reproduce, thanks for reporting.

BFH-ktt1, 2022-08-24:
@MarcinOrlowski : Maybe you have a nice solution to this problem:
In com.cburch.logisim.gui.hex (Hexfile) we have:

`
import static com.cburch.logisim.gui.Strings.S;

....

  public static void save(
....
        final var confirm =
            OptionPane.showConfirmDialog(
                parent,
                S.get("confirmOverwriteMessage", f.getName()),
                S.get("confirmOverwriteTitle"),
                OptionPane.YES_NO_OPTION);
`

which shows the above dialog, also the localisation file gui.properties contains the keys.

In com.cburch.logisim.std.memory (Rom.java) we have:

`
import static com.cburch.logisim.std.Strings.S;
`

and here the above save is used.
As std.properties does not contain the keys I assume that we get the problem described above.

So I assume that the static include in Rom.java overrides the one in Hexfile.java.

Do I make some sense?
I could also put the keys in the std.properties file, but is there not a more elegant way (my knowledge of java is too limited for this)?

MarcinOrlowski, 2022-11-04:
@maehne this is tagged `3.8.0`. Can you please update the target?