logisim-evolution/logisim-evolution #1522: UI Fonts look ugly

SashaPGT, 2022-08-23:
Looks like the UI fonts in Logisim-evolution is not antialiased, which makes it look ugly with jagged edges compared to the original Logisim. Anyone else facing this issue?

[image]

maehne, 2022-08-23:
Please, check whether the option "Use anti aliasing" is set to on in tab "Layout" of the Window -> Preferences dialogue:

[image]

If the option is on and antialiasing is not working, please provide more details, about which exact version/package you are using and on what platform (processor, Linux distribution, X Windows or Wayland?, JDK version).

SashaPGT, 2022-08-23:
[image]

The antialiasing option is already on. Happens to both my laptop (Intel GPU) and desktop (NVIDIA GPU, with proprietary driver 515). I'm using the Flatpak version of Logisim Evolution 3.7.2 on Fedora Workstation (GNOME) 36 on an amd64.

Happens on both X11 and Wayland. The host JDK is openjdk 11.0.16 2022-07-19 (shouldn't matter? since Logisim Evolution is installed via Flatpak)

BFH-ktt1, 2022-08-24:
This could be a flatpack issue, please try to install logisim "native" and see if you have the same problem.

SashaPGT, 2022-08-24:
I tried installing the .jar version and the UI fonts rendered fine, so this is definitely a Flatpak issue.

Also, I tried messing around Flatpak settings (via Flatseal) and found out that after enabling `share=ipc` the UI fonts now render correctly.

BFH-ktt1, 2022-08-24:
I'll try to find out how we can fix this in flatpak, however, with very low priority as I, for the moment, try get as much bugs/issues fixed as possible for the next release, which we plan real soon. Can you live for the moment with the current solution?
Thanks again for the issue and your effort to pin-point the problem!

SashaPGT, 2022-08-25:
I'll try to get that into the [manifest](https://github.com/flathub/com.github.reds.LogisimEvolution/blob/master/com.github.reds.LogisimEvolution.yml)

SashaPGT, 2022-08-25:
flathub/com.github.reds.LogisimEvolution#23

closed 2022-08-25 by SashaPGT